JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #1: Add Helios 300 PH315-52-748u to supported devices

RuiGuilherme, 2021-06-06:
**Kernel**: 5.12.9-arch1-1
**OS**: Arch Linux x86_64
**Device**: Acer Predator Helios 300 PH315-52-748u

1. All mode works fine.
2. Speed control works fine
3. Brightness control works fine (but has one issue with mode).
4. Direction control works fine 
5. cR, cG and cB works fine

I just had one issue: 
If you set keyboard brightness using -b the mode will be set to 3 automatically. (Control brightness using Fn + F9/F10 solve this)

JafarAkhondali, 2021-06-19:
Thank for the repor @RuiGuilherme, glad it worked
 
I understand the problem with brightness, that's because the script is minimal and doesn't store\use the state.
Actually I was hoping to merge the changes into Linux kernel, but some required functionalities for implementations are missing so I'm not sure to wait for them, or develop this repo.

JafarAkhondali, 2021-08-15:
@RuiGuilherme I've added Turbo mode support as well, can you kindly please test updated version? (at your own risk 😆 )

RuiGuilherme, 2021-08-15:
Nice! Apparently works fine and the fans were for maximum speed.

Later I will check if the UnderVolt is being removed and the overclock is being applied.

JafarAkhondali, 2021-08-16:
Probably because it hasn't received the update. I think our CPU was build after the last commit of that repository.
btw I think turbo for helios 300 works only on GPU (not sure)

RuiGuilherme, 2021-08-16:
Hup! You are right. Just overclock GPU, but at the time I bought this notebook it came with Windows, and it had an undervolt on the i7 9750H and Turbo Mode "Disable" this undervolt.

I don't remember the exact values because I bought this notebook in 2020, but I had used this thead to reverse the undervolt without using turbo mode. 
https://www.techpowerup.com/forums/threads/undervolt-an-i7-9750h-with-throttlestop.268380/ 

But it's like I said, maybe I'm going crazy haha

closed 2021-08-25 by JafarAkhondali