JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #2: Worked on PH317-53

inakrin, 2021-08-15:
Thank you so much! 
System Information
        Manufacturer: Acer            
        Product Name: Predator PH317-53
        Version: V1.02
        Serial Number: NHQ5PST00192509D812600
        UUID: 49ddfdd9-4a0a-446a-a154-cee2e382cdaa
        Wake-up Type: Power Switch
--
Base Board Information
        Manufacturer: CFL
        Product Name: Alphard_CFS
        Version: V1.02
        Serial Number: NBQ5P11006924000E42600
        Asset Tag: Type2 - Board Asset Tag
        Features:

JafarAkhondali, 2021-08-15:
@inakrin I'm glad it helped you, thanks for letting me know
Which features did work for you? turbo or LED?

inakrin, 2021-08-16:
@JafarAkhondali both

closed 2021-08-25 by JafarAkhondali