JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #3: Worked for Acer Predator PT315-51

ivan-petrunko, 2021-08-20:
Hey dude! :wave: 

Great job! :muscle:  It worked for my laptop Acer Predator PT315-51 :+1: Both TURBO and RGB modes are working! That's amazing. I waited for it for almost a year since purchasing laptop.
P.S. linux one love :heart: 

Details:
```
System Information
        Manufacturer: Acer
        Product Name: Predator PT315-51
        Version: V1.08
        Serial Number: NHQ6DER0090130A4963400
        UUID: 85c570e7-fa21-ea11-80d3-08979889cc8d
        Wake-up Type: Power Switch
--
Base Board Information
        Manufacturer: CFL
        Product Name: Auris_CFS
        Version: V1.08
        Serial Number: NBQ6D1100301384DC43400
        Asset Tag: Type2 - Board Serial Number
        Features:
```

ivan-petrunko, 2021-08-20:
I have a question about TURBO. Isn't LED light should be on :bulb: when my laptop is in TURBO mode?

JafarAkhondali, 2021-08-25:
Thank Ivan,  Happy to hear it's helping 🥂 
Yes, turbo mode should turn on LED ( it does on Predator Helios series), but maybe the Triton series use a different LED. I'll try to look into that.

ivan-petrunko, 2021-08-25:
@JafarAkhondali , Cool. Thanks a lot :handshake: 
Also I have found out that after rebooting my laptop software works weird, not allowing to manage LED and turn on TURBO. Reinstalling of kernel module helps. 
Have you noticed any problems after rebooting your laptop?

JafarAkhondali, 2021-08-25:
@ivan-petrunko Yes you are right about the problem
As suggested Zehra, here is a temporary fix for problem:
```
sudo ./uninstall.sh
sudo rm /dev/acer-gkbbl-0
sudo ./install.sh
```

I'm really busy but I'll try to fix it as well

ivan-petrunko, 2021-08-25:
@JafarAkhondali , thanks a lot.

closed 2021-08-25 by ivan-petrunko