JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #4: Predator PH315-53 - Install problem

vladimir-djurovic, 2021-08-25:
I follow instructions from README.md and on installation a got this error

My model **Predator PH315-53**

[image]

JafarAkhondali, 2021-08-25:
@vladimir-djurovic Hi
Please do the following, and let me know the result:
1. run `sudo apt update && sudo apt install linux-headers-generic` ( if not Debian based, use your own distro package manager)
2. instead of `sh`, use `bash` to run the installer

vladimir-djurovic, 2021-08-25:
Hi @JafarAkhondali 

Thanks for the reply but I still get the same problem.

I use **Linux Mint 20.2 Cinnamon**, kernel **5.4.0-81-generic** and cinnamon version **5.0.5**

vladimir-djurovic, 2021-08-25:
Ty @JafarAkhondali,

Turbo Mode is working, but the RGB light still not working.

I tried the following:
[image]

,and
[image]

JafarAkhondali, 2021-08-25:
try:
```
sudo ./uninstall.sh
sudo rm /dev/acer-gkbbl-0
sudo ./install.sh
```

and make sure secure boot is disabled in bios.

which of the previous steps did fix the turbo mode problem for you? I wanna add them to code

vladimir-djurovic, 2021-08-25:
Now works, @JafarAkhondali thank you so much.

To fix the turbo mode, I had to include both header files.

closed 2021-09-05 by JafarAkhondali

alok-37, 2022-07-26:
Would have been really helpful if this worked on PH315-55.I did try to install it.It either works occasionally or it doesn't.

fjmarcelo, 2023-11-24:
> try:
> 
> ```
> sudo ./uninstall.sh
> sudo rm /dev/acer-gkbbl-0
> sudo ./install.sh
> ```
> 
> and make sure secure boot is disabled in bios.
> 
> which of the previous steps did fix the turbo mode problem for you? I wanna add them to code

Same problem, but I can't fix it.

I'm using `Arch Linux` and `Predator PHN16-71`.

What can I do? Thanks in advance.

Order52, 2024-11-02:
It doesn't work on my Helios 16 PH16-71 it changes nothing when I configure the rgb settings is there a way to this?

Sloppy@bazzite:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo dmidecode | grep "Product Name" -B 2 -A 4
System Information
	Manufacturer: Acer
	Product Name: Predator PH16-71
	Version: V1.16
	Serial Number: NHQJQAA002325186517600
	UUID: ee6a13fd-c099-6445-bf04-74d4dd0896ab
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: RPL
	Product Name: Discovery_RTX
	Version: V1.16
	Serial Number: NBQJQ11001325186517600
	Asset Tag: Type2 - Board Asset Tag
	Features:
Sloppy@bazzite:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.aml
Sloppy@bazzite:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ 

[image]