JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #5: Support for Predator Triton 300 SE

mjkapkan, 2021-09-01:
Backlight control working, but everything else not.

Installation log:
```
make -C /lib/modules/5.11.0-27-generic/build M=/temp/acer-helios-300-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-27-generic'
  CC [M]  /temp/acer-helios-300-rgb-keyboard-linux-module/src/facer.o
  MODPOST /temp/acer-helios-300-rgb-keyboard-linux-module/Module.symvers
  CC [M]  /temp/acer-helios-300-rgb-keyboard-linux-module/src/facer.mod.o
  LD [M]  /temp/acer-helios-300-rgb-keyboard-linux-module/src/facer.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-27-generic'
[27481.787691] acer_wmi: Unknown function number - 4 - 0
[27481.975611] acer_wmi: Unknown function number - 4 - 0
[27482.136116] acer_wmi: Unknown function number - 4 - 0
[27482.311916] acer_wmi: Unknown function number - 4 - 0
[27482.447871] acer_wmi: Unknown function number - 4 - 0
[27568.986448] acer_wmi: Acer Laptop WMI Extras unloaded
[27569.023806] facer: module verification failed: signature and/or required key missing - tainting kernel
[27569.024464] facer: Acer Laptop ACPI-WMI Extras
[27569.024506] facer: Function bitmap for Communication Button: 0x801
[27569.039239] input: Acer WMI hotkeys as /devices/virtual/input/input53
[*] Done

```

Output of `dmidecode | grep "Product Name" -B 2 -A 4`
```
System Information
	Manufacturer: Acer
	Product Name: Predator PT314-51s
	Version: V1.06
	Serial Number: NHQBJEL003123043633400
	UUID: 56cd3b12-16c5-eb11-85af-088fc3004591
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: TGL
	Product Name: Clubman_TLM
	Version: V1.06
	Serial Number: NBQBJ11005123BE4813400
	Asset Tag: Type2 - Board Asset Tag
	Features:

```
[dsdt.aml.txt](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/7091624/dsdt.aml.txt)

JafarAkhondali, 2021-09-01:
Hi,
judging by logs, I think the key nums of your laptop for turbo button differs from mine.

can you apply these changes, the uninstall & install again?

in src/facer.c:
Replace line 94:
`	WMID_GAMING_TURBO_KEY_EVENT = 0x7,`
with: 
`	WMID_GAMING_TURBO_KEY_EVENT = 0x4,`

and replace line 2428:
`			if (return_value.key_num == 0x4)`
with:
`			if (return_value.key_num == 0x0)`

mjkapkan, 2021-09-01:
Hi, thanks for the support.
Tried it out. Different messages during installation, but the behavior is unchanged.
```
[32409.578716] R13: 00007fffca501782 R14: 0000560cdc8f52a0 R15: 0000560cdc8f5780
[32409.578722] ---[ end trace 2a89544a8aff7104 ]---
[32409.578963] facer: Acer Laptop WMI Extras unloaded
[32409.611412] acer_wmi: Acer Laptop ACPI-WMI Extras
[32409.611440] acer_wmi: Function bitmap for Communication Button: 0x801
[32409.624818] input: Acer WMI hotkeys as /devices/virtual/input/input64
[32417.038080] acer_wmi: Acer Laptop WMI Extras unloaded
[32417.062596] facer: Acer Laptop ACPI-WMI Extras
[32417.062622] facer: Function bitmap for Communication Button: 0x801

```e
[dsdt.aml.txt](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/7092351/dsdt.aml.txt)

JafarAkhondali, 2021-09-01:
Actually the patch currently is not supported with my list. There can be a fix but will need to be tested ...
here is a try (use at your own risk):

in line 756, add this:
```
		{
				.callback = dmi_matched,
				.ident = "Acer Predator PT314-51s",
				.matches = {
						DMI_MATCH(DMI_SYS_VENDOR, "Acer"),
						DMI_MATCH(DMI_PRODUCT_NAME, "Predator PT314-51s"),
				},
				.driver_data = &quirk_acer_predator_pt314_51s,
		},
```

then at line 462, add this:
```
static struct quirk_entry quirk_acer_predator_pt314_51s = {
		.turbo = 1,
		.cpu_fans = 1,
		.gpu_fans = 1,
};
```

I think your laptop comes with 1 fan for cpu and 1 fan for gpu, like mine. If these are not correct, change the number accordingly.

If you tested these and didn't work for you, please give me the link of the predator app you are using

JafarAkhondali, 2021-09-01:
@mjkapkan Ok, I'm a little busy and can't spare time for it now, but I'll keep issue open in case someone wanted to de-compile the official app and add the feature.

mjkapkan, 2021-09-01:
Thanks for the help so far. I will try to look in to this myself once I have some spare time.

NikitaGolovaniuk, 2022-04-14:
> Thanks for the help so far. I will try to look in to this myself once I have some spare time.

Did you make it work? I spent some time to get work registers, even turbo button light but i cant find keycode of turbo button

johandroid, 2023-07-22:
Hello everyone, 

Do you have any progress to share? I have a PT314-52S and is heating a lot. Can you help me with this?
Thank you very much.