JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #9: Not working on my PH315-52

Leonardo-Cavalcante, 2021-09-03:
Hello! 

I've tried installing it on my laptop, but it didn't work. There was no error message while installing, nor when trying to change the keyboard backlight, however, it does not change it. Also, when i press the turbo button, nothing happens.

These are the outputs you requested in the readme.md:

```
System Information
        Manufacturer: Acer
        Product Name: Predator PH315-52
        Version: V1.12
        Serial Number: NHQ63AL001020DED369501
        UUID: 99e70168-dad9-40eb-835b-03b1e78a5419
        Wake-up Type: Power Switch
--
Base Board Information
        Manufacturer: CFL
        Product Name: Covini_CFS
        Version: V1.12
        Serial Number: NBQ5411004014024A99501
        Asset Tag: Type2 - Board Asset Tag
        Features:

```

[dsdt.aml.txt](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/7107178/dsdt.aml.txt)

JafarAkhondali, 2021-09-03:
Hi
Can you run these:
```
sudo ./uninstall.sh
sudo rm /dev/acer-gkbbl-0
sudo ./install.sh
```

and atack full log here as well?

Leonardo-Cavalcante, 2021-09-05:
Hey
Now it shows up this error:

```
make -C /lib/modules/5.13.12-1-MANJARO/build M=/home/leo/acer-helios-300-rgb-keyboard-linux-module modules
make[1]: *** /lib/modules/5.13.12-1-MANJARO/build: No such file or directory.  Stop.
make: *** [Makefile:9: default] Error 2
rmmod: ERROR: Module acer_wmi is not currently loaded
insmod: ERROR: could not load module src/facer.ko: No such file or directory
[  241.065781] alx 0000:07:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  241.065788] alx 0000:07:00.0:   device [1969:e0b1] error status/mask=00000040/00002000
[  241.065794] alx 0000:07:00.0:    [ 6] BadTLP                
[  241.066436] pcieport 0000:00:1d.4: AER: Multiple Corrected error received: 0000:07:00.0
[  241.066702] alx 0000:07:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  241.066708] alx 0000:07:00.0:   device [1969:e0b1] error status/mask=00000080/00002000
[  241.066713] alx 0000:07:00.0:    [ 7] BadDLLP               
[  248.300571] audit: type=1101 audit(1630849819.943:233): pid=3082 uid=1000 auid=1000 ses=2 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="leo" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  248.301106] audit: type=1110 audit(1630849819.943:234): pid=3082 uid=1000 auid=1000 ses=2 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  248.307278] audit: type=1105 audit(1630849819.950:235): pid=3082 uid=1000 auid=1000 ses=2 subj==unconfined msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[*] Done

```

JafarAkhondali, 2021-09-05:
try `sudo pacman -S linux-headers` or `sudo pacman -S linux-headers-$(uname -r)` then former commands again.

Leonardo-Cavalcante, 2021-09-05:
Thanks! Now the turbo works, but no rgb though. Here's the log.

[dsdt.aml.txt](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/7111751/dsdt.aml.txt)

Leonardo-Cavalcante, 2021-09-05:
I think i messed up LOL, now it works after running these again

```
sudo ./uninstall.sh
sudo rm /dev/acer-gkbbl-0
sudo ./install.sh

```
Thanks a lot!

closed 2021-09-05 by Leonardo-Cavalcante