JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #10: [Acer Nitro AN515-45] Works well!

SanyStation, 2021-09-12:
`sudo dmidecode | grep "Product Name" -B 2 -A 4`

 System Information
        Manufacturer: Acer
        Product Name: Nitro AN515-45
        Version: V1.03
        Serial Number: NHQBREU00A1130D15F3400
        UUID: 5278f73b-5990-eb11-80dc-089798eb9293
        Wake-up Type: Power Switch

Base Board Information
        Manufacturer: CZ
        Product Name: Scala_CAS
        Version: V1.03
        Serial Number: NBQBR110021136F1433400
        Asset Tag: Type2 - Board Asset Tag
        Features:

Thank you!

SanyStation, 2021-09-12:
Closing the issue because it's not an issue

closed 2021-09-12 by SanyStation

JafarAkhondali, 2021-09-12:
@SanyStation Thanks for reporting!
Did turbo mode worked for you too?

SanyStation, 2021-09-12:
I'm not sure whether I have such a mode. I don't really have a 'turbo' button on the keyboard. Does Acer Nitro provide this mode? How to activate it? What should it exactly be?

stefanosiano, 2021-09-16:
I think turbo button is not present on Nitro series (I'm not too sure, but I've never seen it). Instead there is a dedicated button to open the NitroSense app.
Love this project and hope it will work on all Nitro products!
@JafarAkhondali Thanks for your efforts!
Maybe add AN515-45 to the compatibility list? Also a typo in the same list: A515-55 -> AN515-55

JafarAkhondali, 2021-09-16:
Oh thanks I get it now, there is no dedicated button for turbo
I'll fix typo and add new model as well, thanks