JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #11: acer nitro 5 AN17-41 working

stefanosiano, 2021-09-17:
Can confirm it works on Acer Nitro 5 AN517-41!

And a note about static coloring. Here is an ugly hack to make it work (single color, through):
#this changes the mode so that when changing to Zoom mode animation starts from beginning
python3 facer_rgb.py -m1
#set Zoom mode with speed 9
python3 facer_rgb.py -m5 -s9
#wait until animation makes the whole keyboard colored (sleep time could be a parameter for better compatibility?)
sleep 0.3
#stop zoom animation and set the color
python3 facer_rgb.py -m 5 -s 0 -cR 0 -cB 0 -cG 255

Maybe you could add it to your python script as a new mode (static single color) until you find a way to make real static coloring working?

Thanks for your work anyway @JafarAkhondali!
  
sudo dmidecode | grep "Product Name" -B 2 -A 4
System Information
	Manufacturer: Acer
	Product Name: Nitro AN517-41
	Version: V1.08
	Serial Number: NHQARET0051270120F3400
	UUID: dcc67398-0fd7-eb11-80de-088fc306bd6d
	Wake-up Type: Power Switch

Base Board Information
	Manufacturer: CZ
	Product Name: Kamiq_CAS
	Version: V1.08
	Serial Number: NBQBC110021278A9C13400
	Asset Tag: Type2 - Board Asset Tag

JafarAkhondali, 2021-09-18:
Hi,
I'll add this to compatibility list and also implement static coloring soon.
I did similar workaround for static coloring but since it's not customizable I will implement static coloring, and if it didn't work, I'll add your workaround.
Thanks

stefanosiano, 2021-09-20:
Great!
I also noticed a weird thing after using your python script.
When I press the buttons on my keyboard to change the keyboard brightness, it changes the keyboard brightness as expected, but it also drops the laptop brightness to the minimum level.
It happens only the very first time I run a command to change the keyboard rgb mode. After that, everything works fine.
Noticed it only happens after running the python command. If I simply install the module with the install script, this doesn't happen, unless I run the python script.
Don't know if it's of any help, but my laptop is an Amd/Nvidia model.
Hope it helps.

Thanks for this module! :)

JafarAkhondali, 2021-09-22:
You are right about the bug. I haven't yet detected the problem.
Maybe I didn't identified correct bytes on WMI call.
Thanks

JafarAkhondali, 2021-09-23:
Thanks to https://www.spinics.net/lists/linux-leds/msg18741.html, this commit https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/commit/30804d316fbe416961fb61157f7462423e027f0d adds support for static coloring mode

closed 2021-09-23 by JafarAkhondali

stefanosiano, 2021-09-23:
Just tried it
I ran:
```
sudo uninstall.sh
sudo rm /dev/acer-gkbbl-0 #just for safety
cd ~
git clone https://github.com/JafarAkhondali/acer-helios-300-rgb-keyboard-linux-module
cd "acer-helios-300-rgb-keyboard-linux-module"
chmod +x ./install.sh
sudo ./install.sh
```
I even restarted my laptop, but the mode "0" doesn't work. It just turns off the keyboard lights. All other modes work as expected, like before.
Should I have waited a little before trying? :D

JafarAkhondali, 2021-09-23:
well, it works on my machine 🤣 
Can you try this:
`for i in $(seq 1 4); do python3 facer_rgb.py -m 0 -z $i -cR 255 -cB 0 -cG 0 -b 100; done`

check if "/dev/acer-gkbbl-static-0" exists
check install output
and check `sudo dmesg | tail -n 50`

stefanosiano, 2021-09-23:
Still nothing.
Here are the outputs

```
sudo sh install.sh 
make -C /lib/modules/5.13.16-200.fc34.x86_64/build M=/home/stefano/acer-helios-300-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/kernels/5.13.16-200.fc34.x86_64'
make[1]: Leaving directory '/usr/src/kernels/5.13.16-200.fc34.x86_64'
[   12.346942] wlp4s0: Limiting TX power to 27 (30 - 3) dBm as advertised by f0:b0:14:79:76:fc
[   12.834351] IPv6: ADDRCONF(NETDEV_CHANGE): wlp4s0: link becomes ready
[   13.043401] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   14.396212] rfkill: input handler disabled
[   19.827094] rfkill: input handler enabled
[   22.294363] rfkill: input handler disabled
[  150.778890] acer_wmi: Acer Laptop WMI Extras unloaded
[  150.802192] facer: Acer Laptop ACPI-WMI Extras
[  150.802220] facer: Function bitmap for Communication Button: 0x801
[  150.815452] input: Acer WMI hotkeys as /devices/virtual/input/input27
[*] Done
```

```
file /dev/acer-gkbbl-static-0
/dev/acer-gkbbl-static-0: character special (506/0)
```

```
sudo dmesg | tail -n 50
[    4.998263] snd_hda_codec_realtek hdaudioC1D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.998264] snd_hda_codec_realtek hdaudioC1D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    4.998265] snd_hda_codec_realtek hdaudioC1D0:    mono: mono_out=0x0
[    4.998266] snd_hda_codec_realtek hdaudioC1D0:    inputs:
[    4.998267] snd_hda_codec_realtek hdaudioC1D0:      Mic=0x12
[    5.101877] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 1
[    5.240188] input: HD-Audio Generic Headphone as /devices/pci0000:00/0000:00:08.1/0000:05:00.6/sound/card1/input26
[    5.438453] kauditd_printk_skb: 99 callbacks suppressed
[    5.438460] audit: type=1130 audit(1632418619.430:108): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-udev-settle comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.457403] BTRFS info (device sda): enabling ssd optimizations
[    5.457406] BTRFS info (device sda): disk space caching is enabled
[    5.457407] BTRFS info (device sda): has skinny extents
[    5.468961] audit: type=1130 audit(1632418619.461:109): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-fsck@dev-disk-by\x2duuid-4aa4f11f\x2ded3f\x2d4f00\x2d82bd\x2dc3eefdf3fb8d comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.470636] audit: type=1130 audit(1632418619.463:110): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-fsck@dev-disk-by\x2duuid-BECA\x2dC03E comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.473444] EXT4-fs (nvme0n1p2): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[    5.490049] audit: type=1130 audit(1632418619.482:111): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=dracut-shutdown comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.502518] audit: type=1130 audit(1632418619.494:112): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=plymouth-read-write comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.504758] audit: type=1130 audit(1632418619.497:113): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=import-state comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.547827] audit: type=1130 audit(1632418619.540:114): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-tmpfiles-setup comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.551622] audit: type=1334 audit(1632418619.544:115): prog-id=26 op=LOAD
[    5.551994] audit: type=1334 audit(1632418619.544:116): prog-id=27 op=LOAD
[    5.552142] audit: type=1334 audit(1632418619.544:117): prog-id=28 op=LOAD
[    5.596755] RPC: Registered named UNIX socket transport module.
[    5.596758] RPC: Registered udp transport module.
[    5.596758] RPC: Registered tcp transport module.
[    5.596759] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    5.697433] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    5.697440] Bluetooth: BNEP filters: protocol multicast
[    5.697447] Bluetooth: BNEP socket layer initialized
[    5.727923] loop: module loaded
[    5.755064] loop0: detected capacity change from 0 to 8
[    5.775609] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    6.327532] Generic FE-GE Realtek PHY r8169-0-300:00: attached PHY driver (mii_bus:phy_addr=r8169-0-300:00, irq=MAC)
[    6.488746] r8169 0000:03:00.0 enp3s0: Link is Down
[   11.735564] wlp4s0: authenticate with f0:b0:14:79:76:fc
[   11.744615] wlp4s0: send auth to f0:b0:14:79:76:fc (try 1/3)
[   11.747588] wlp4s0: authenticated
[   11.753548] wlp4s0: associate with f0:b0:14:79:76:fc (try 1/3)
[   11.756627] wlp4s0: RX AssocResp from f0:b0:14:79:76:fc (capab=0x1511 status=0 aid=3)
[   12.262361] wlp4s0: associated
[   12.346942] wlp4s0: Limiting TX power to 27 (30 - 3) dBm as advertised by f0:b0:14:79:76:fc
[   12.834351] IPv6: ADDRCONF(NETDEV_CHANGE): wlp4s0: link becomes ready
[   13.043401] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   14.396212] rfkill: input handler disabled
[   19.827094] rfkill: input handler enabled
[   22.294363] rfkill: input handler disabled
[  150.778890] acer_wmi: Acer Laptop WMI Extras unloaded
[  150.802192] facer: Acer Laptop ACPI-WMI Extras
[  150.802220] facer: Function bitmap for Communication Button: 0x801
[  150.815452] input: Acer WMI hotkeys as /devices/virtual/input/input27

```

commit 30804d31 (2021-09-23): Implement static coloring mode
Fixes #11