JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #12: Cannot change RGB light on my ACER Nitro AN515-55

GamerMine, 2021-09-18:
Hello,
After installing the module, I try to use one of the templates and it does not work. It's not changing anything on my keyboard.
I use Ubuntu 21.04. The linux header is installed.
I hope I can use this module =)

JafarAkhondali, 2021-09-18:
Hi
Did you disable secure boot in bios?
As reported by https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/7, it should work for your laptop.

GamerMine, 2021-09-18:
Thank you for your response =)
Yes the Secure Boot is disabled in bios =)

JafarAkhondali, 2021-09-18:
Can you try these steps as well?

`sudo ./uninstall.sh`
`sudo apt update && sudo apt install linux-headers-generic linux-headers-$(uname -r)`

add `#include <linux/uaccess.h>` and  `#include <asm/uaccess.h>` after line 32 of src/facer.c ( first try to add them separately, if none worked, try to add them both)

```
sudo rm /dev/acer-gkbbl-0
sudo ./install.sh
```

and try running python script with sudo. It should work then. If not, check the output of ./install.sh command and `dmesg`

GamerMine, 2021-09-18:
So I tried everything but nothing works. Here is the output of sudo ./install.sh : 
`make -C /lib/modules/5.11.0-18-generic/build M=/home/maxime/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.11.0-18-generic »
  CC [M]  /home/maxime/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
  MODPOST /home/maxime/acer-predator-turbo-and-rgb-keyboard-linux-module/Module.symvers
  CC [M]  /home/maxime/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.mod.o
  LD [M]  /home/maxime/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
  BTF [M] /home/maxime/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
Skipping BTF generation for /home/maxime/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko due to unavailability of vmlinux
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.11.0-18-generic »
[  992.679550] facer: Function bitmap for Communication Button: 0x800
[  992.710315] input: Acer WMI hotkeys as /devices/virtual/input/input34
[ 1009.845259] facer: Acer Laptop WMI Extras unloaded
[ 1009.880076] acer_wmi: Acer Laptop ACPI-WMI Extras
[ 1009.880103] acer_wmi: Function bitmap for Communication Button: 0x800
[ 1009.892500] input: Acer WMI hotkeys as /devices/virtual/input/input35
[ 1080.460885] acer_wmi: Acer Laptop WMI Extras unloaded
[ 1080.502834] facer: Acer Laptop ACPI-WMI Extras
[ 1080.502877] facer: Function bitmap for Communication Button: 0x800
[ 1080.517286] input: Acer WMI hotkeys as /devices/virtual/input/input36
[*] Done
`
And here is the last few lines of the output of dmesg : 
`[ 1080.460885] acer_wmi: Acer Laptop WMI Extras unloaded
[ 1080.502834] facer: Acer Laptop ACPI-WMI Extras
[ 1080.502877] facer: Function bitmap for Communication Button: 0x800
[ 1080.517286] input: Acer WMI hotkeys as /devices/virtual/input/input36
`

GamerMine, 2021-09-18:
So I tried everything but it is still nor working. Here is the output of `sudo ./install.sh` : 
[image]

And there is the last few lines of the output of `dmesg` : 
[image]

JafarAkhondali, 2021-09-18:
I think this has something to do with vmlinux.

try these
```
echo "deb http://ddebs.ubuntu.com $(lsb_release -cs) main restricted universe multiverse" | sudo tee -a /etc/apt/sources.list.d/ddebs.list
echo "deb http://ddebs.ubuntu.com $(lsb_release -cs)-updates main restricted universe multiverse"  | sudo tee -a /etc/apt/sources.list.d/ddebs.list
apt-get update
sudo apt-get install linux-image-$(uname -r)-dbgsym
```

Source: https://www.ibm.com/docs/en/linux-on-systems?topic=linuxonibm/liacf/oprofkernelsymrhel.html

GamerMine, 2021-09-18:
Okay so I have installed the vmlinux thing and it is not complaining about it anymore. But it is still not working =)
The `sudo ./install.sh` output as slightly changed so here is it : 
[image]

JafarAkhondali, 2021-09-18:
hmm, try installing other kernels for your distro. I am pretty sure that this should work with your laptop, it's just a software problem

GamerMine, 2021-09-19:
I updated the kernel of my laptop and it is still not working. My previous kernel was 5.11.0-18 and now it is 5.11.0-34.
When recompiling (when I add or remove the two includes) it always complain about vmlinux.
Is it normal that in dmseg it says : `acer_wmi: Acer Laptop WMI Extras unloaded` ?

JafarAkhondali, 2021-09-19:
Then give Ubuntu live a try. Unloading acer-wmi is normal, cause my repo is the extended version of acer-wmi

GamerMine, 2021-09-19:
Okay, so I tried in an Ubuntu live and it doesn't work either. I start thinking that my computer is not compatible =( Maybe I have a slightly different revision of the Nitro AN515-55 ? I really don't know

JafarAkhondali, 2021-09-19:
If you have a 4-zone RGB keyboard, then this should work for you

GamerMine, 2021-09-19:
Okay after a lot a search I found that the AN515-55 model has a lot of different revision and I figured out that if the keyboard has a red painting when it is off, then the keyboard is not RGB and unfortunately my keyboard have that paint.
I'm sorry for all of this =)

JafarAkhondali, 2021-09-19:
😄 Ok

closed 2021-09-19 by JafarAkhondali