JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #14: Lost support on Acer Predator PH315-53

lucasrueda, 2021-09-19:
It was working for me but not anymore!

I think I updated my Ubuntu 20.04 but I'm not sure.

```
lrueda@lrueda-predator:~/.acer-300-linux$ sudo ./install.sh 
make -C /lib/modules/5.11.0-34-generic/build M=/home/lrueda/.acer-300-linux modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-34-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-34-generic'
rmmod: ERROR: Module acer_wmi is not currently loaded
insmod: ERROR: could not insert module src/facer.ko: Invalid module format
[ 4829.265895] systemd[1]: systemd-journald.service: Succeeded.
[ 4829.266219] systemd[1]: Stopped Journal Service.
[ 4829.268010] systemd[1]: Starting Journal Service...
[ 4829.298812] systemd[1]: Started Journal Service.
[ 4829.304836] systemd-journald[31812]: Received client request to flush runtime journal.
[ 4830.438110] audit: type=1400 audit(1632078726.994:162): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="virt-aa-helper" pid=31990 comm="apparmor_parser"
[ 4830.487659] audit: type=1400 audit(1632078727.042:163): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="libvirtd" pid=31993 comm="apparmor_parser"
[ 4830.514688] audit: type=1400 audit(1632078727.070:164): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="libvirtd//qemu_bridge_helper" pid=31993 comm="apparmor_parser"
[ 4831.756341] audit: type=1400 audit(1632078728.316:165): apparmor="DENIED" operation="capable" profile="libvirtd" pid=32071 comm="libvirtd" capability=17  capname="sys_rawio"
[ 4860.488928] facer: disagrees about version of symbol module_layout
[*] Done
lrueda@lrueda-predator:~/.acer-300-linux$
```

lucasrueda, 2021-09-19:
Well I don't know what happened but I tried to clone the repo again.. and it worked.

closed 2021-09-19 by lucasrueda

JafarAkhondali, 2021-09-19:
👀 
[image]