JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #17: Failed install on PH315-54

Soothsayer86, 2021-09-23:
Please excuse any formatting issue as this is my first GitHub Post
I've recently install Ubuntu 20.04.3 LTS 64-bit GNOME 3.36.8
1. I did locate the sys folder 7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56
2. Everything went smooth until the below "sudo ./unstall.sh

dfresh@dfresh:~/acer-helios-300-rgb-keyboard-linux-module$ sudo ./install.sh
./install.sh: line 17: make: command not found
insmod: ERROR: could not load module src/facer.ko: No such file or directory
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   30.829208] audit: type=1107 audit(1632433331.624:44): pid=708 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.10" pid=2141 label="snap.snap-store.ubuntu-software" peer_pid=722 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   30.829303] audit: type=1107 audit(1632433331.624:45): pid=708 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="CheckAuthorization" mask="send" name=":1.10" pid=2141 label="snap.snap-store.ubuntu-software" peer_pid=722 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   31.031103] audit: type=1400 audit(1632433331.824:46): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/etc/PackageKit/Vendor.conf" pid=2141 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   36.069672] audit: type=1400 audit(1632433336.740:47): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/gdm/greeter/applications/gnome-initial-setup.desktop" pid=2141 comm="pool-org.gnome." requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   36.082135] audit: type=1400 audit(1632433336.752:48): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/gdm/greeter/applications/gnome-initial-setup.desktop" pid=2141 comm="pool-org.gnome." requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   36.404985] audit: type=1326 audit(1632433337.064:49): auid=1000 uid=1000 gid=1000 ses=5 subj=snap.snap-store.ubuntu-software pid=2141 comm="pool-org.gnome." exe="/snap/snap-store/547/usr/bin/snap-store" sig=0 arch=c000003e syscall=93 compat=0 ip=0x7f1d950214e7 code=0x50000
[  464.198403] acer_wmi: Acer Laptop WMI Extras unloaded
[*] Done

Any help is much appreciated!

JafarAkhondali, 2021-09-23:
Did you disable secure boot? it must be disabled
If that didn't work, try to disable apparmor

Shawshank-dulal, 2022-03-21:
I am sorry that I forgot to reply. I have disabled secure boot, tried disabling apparmor and so on. It is not working.
Here is the output of the installation:
[image]

and Here is the output when I try to run facer_rgb.py
[image]

mmsaeed509, 2022-03-21:
@Shawshank-dulal 
try 
```bash
sudo rm /dev/acer-gkbbl-0
sudo./uninstal_service.sh
sudo apt install linux-headers
sudo ./install_service.sh
```

mmsaeed509, 2022-03-21:
@Shawshank-dulal  try
```
sudo apt install linux-headers
sudo ./install_service.sh
```

Shawshank-dulal, 2022-03-21:
@mmsaeed509 Thanks for your quick reply. I have already tried that. It is not working even after uninstalling and reinstalling.

mmsaeed509, 2022-03-21:
the `acer-gkbbl-0` file is not the correct file 

in my machine is `/dev/acer-gkbbl-static-0`   use the TAB key to complete the correct file 

I encountered this problem, here is my solution

```
sudo rm /dev/acer-gkbbl-static-0

sudo ./uninstall_service.sh

cd ..

rm -rf acer-predator-turbo-and-rgb-keyboard-linux-module

sudo pacman -S linux-headers 

git clone https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module.git

cd acer-predator-turbo-and-rgb-keyboard-linux-module

chmod +x install_service.sh

```
then install the module

`sudo ./install_service.sh`

and her is the out put
```
[Mode: install]
[Remove old data]
[Create directories]
[Copy new data]
sending incremental file list
LICENSE
Makefile
README.md
facer_rgb.py
install.sh
install_openrc.sh
install_service.sh
keyboard.webp
refresh.sh
uninstall.sh
uninstall_service.sh
src/
src/facer.c

sent 2,644,072 bytes  received 256 bytes  5,288,656.00 bytes/sec
total size is 2,642,584  speedup is 1.00
[Create turbo-fan service]
Created symlink /etc/systemd/system/multi-user.target.wants/turbo-fan.service → /etc/systemd/system/turbo-fan.service.
```
worked well with me, try it may be working with you

Shawshank-dulal, 2022-03-21:
Thank you for your effort. It is still not working since the install is producing different output than you. Please have a look at below:

[Mode: install]
[Remove old data]
[Create directories]
[Copy new data]
sending incremental file list
LICENSE
Makefile
README.md
facer_rgb.py
install.sh
install_openrc.sh
install_service.sh
keyboard.webp
refresh.sh
uninstall.sh
uninstall_service.sh
src/
src/facer.c

sent 2,644,089 bytes  received 256 bytes  5,288,690.00 bytes/sec
total size is 2,642,584  speedup is 1.00
[Create turbo-fan service]
Synchronizing state of turbo-fan.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable turbo-fan
update-rc.d: error: turbo-fan Default-Start contains no runlevels, aborting.

mmsaeed509, 2022-03-21:
[image]

try to give permission `cd /dev && sudo chmod +x acer-gkbbl-0 acer-gkbbl-static-0`