JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #18: Module works on Acer Nitro AN517-41

FergusonRichard, 2021-09-24:
Thank you for coding efforts.

[product_dsdt.zip](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/7228181/product_dsdt.zip)

JafarAkhondali, 2021-09-25:
Thanks, did you test static coloring mode as well?

FergusonRichard, 2021-09-25:
Yes. I have tested all the examples included in the readme (Breath, Neon, Wave, Shifting, Zoom, Static waving, and both Static mode coloring) . I also tested various custom 'Static mode coloring'  RGB settings. They also persist after reboot/power off under the Manjaro Linux 5.14 kernel.

Each zones color is retained and requires being individually set as designed. Piping additional python3 to set other zones simultaneously works in bash, as do scripts with individual line entries.

I am actually using Static mode coloring while writing this comment :)
It seems a nice GUI wrapper would be your next step.

commit c0f99438 (2021-09-26): Update README.md

Fixes #18
closed 2021-09-26 by JafarAkhondali