JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #19: Error in install

jazar17, 2021-10-04:
Hi I have the errors below when installing:

make -C /lib/modules/5.14.8-arch1-1/build M=/home/rud31mp/acer-helios-300-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/5.14.8-arch1-1/build'
make[1]: Leaving directory '/usr/lib/modules/5.14.8-arch1-1/build'
[ 2861.946023] RIP: 0033:0x7f1101b7118d
[ 2861.946030] Code: b4 0c 00 0f 05 eb a9 66 0f 1f 44 00 00 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d b3 6c 0c 00 f7 d8 64 89 01 48
[ 2861.946034] RSP: 002b:00007ffc890f9478 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[ 2861.946042] RAX: ffffffffffffffda RBX: 000055f7c22007b0 RCX: 00007f1101b7118d
[ 2861.946046] RDX: 0000000000000000 RSI: 000055f7c1906270 RDI: 0000000000000003
[ 2861.946050] RBP: 0000000000000000 R08: 0000000000000000 R09: 00007f1101c3c380
[ 2861.946053] R10: 0000000000000003 R11: 0000000000000246 R12: 000055f7c1906270
[ 2861.946056] R13: 0000000000000000 R14: 000055f7c2200760 R15: 0000000000000000
[ 2861.946063] kobject_add_internal failed for acer-gkbbl with -EEXIST, don't try to register things with the same name in the same directory.
[ 2861.970755] input: Acer WMI hotkeys as /devices/virtual/input/input47
[*] Done

then when checking I only have the /dev/acer-gkbbl-static-0 file and not the other one so running the python file causes error. Can you help?

JafarAkhondali, 2021-10-04:
Some times I have this problem after re-installing, but a reboot solves it. Can you test remove `/dev/acer-gkbbl-static-0` and try reboot, then install again?

jazar17, 2021-10-05:
it didn't exactly work, when I change the colors to static the keyboard lights just turns off completely

JafarAkhondali, 2021-10-10:
May I know your exact laptop model? No one reported this problem before. Does static coloring works for you in windows fine? Did you try increasing backlight brightness?
`python3 facer_rgb.py -m 0 -z 4 -cR 255 -cB 255 -cG 0`
What does this command do to your keyboard?

ivan-petrunko, 2021-12-08:
@JafarAkhondali , I have got the same message on my Acer Predator PT315-51:
```
kobject_add_internal failed for acer-gkbbl with -EEXIST, don't try to register things with the same name in the same directory.
```

Cannot manage RGB keyboard backlit. It says:
```
Traceback (most recent call last):
  File "./facer_rgb.py", line 160, in <module>
    with open(CHARACTER_DEVICE, 'wb') as cd:
PermissionError: [Errno 13] Permission denied: '/dev/acer-gkbbl-0'
```

File `/dev/acer-gkbbl-0` is missing. Only `/dev/acer-gkbbl-static-0` present.
```
$ ls -alh /dev/acer-gkbbl-0
ls: cannot access '/dev/acer-gkbbl-0': No such file or directory
```

JafarAkhondali, 2021-12-08:
I also get this error when re-installing the module, did you try installing it as a service (newly added) or manually?  
I don't know the cause of it yet, would be happy if someone can contribute.

ivan-petrunko, 2021-12-08:
@JafarAkhondali , nope, I installed it manually, not as a service.

JafarAkhondali, 2021-12-08:
then reboot, and install it as a service, and check if it's working or not

pranav2305, 2021-12-13:
I have an acer predator helios 300 (PH315-53). I did face some issues while installing but rebooting the laptop solved it. However, after installation the static mode is not working. When i change it to static mode, the keyboard light gets switched off. Please let me know if you find any solution for this.

Anyways, it is a great package @JafarAkhondali. The turbo mode and other lighting modes work flawlessly.

axtloss, 2021-12-31:
I have an acer nitro 5 (AN515-55), static coloring also just turns off the keyboard lights for me, every other color method works perfectly.
Let me know if you need any extra information like system logs and I'll try to provide them

IllusionMan1212, 2023-10-06:
Should be fixed by #121

closed 2023-10-07 by JafarAkhondali