JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #20: Predator G9-793 - POP OS 21.04

kroaton, 2021-10-10:
RGB seems to be working but the default turbo button (pressing red button 1) does nothing.

Product name: Predator G9-793

sudo dmidecode | grep "Product Name" -B 2 -A 4

```
System Information
	Manufacturer: Acer
	Product Name: Predator G9-793
	Version: V1.01
	Serial Number: NHQ17EX005638089677200
	UUID: c3a94130-8423-81e6-31cf-88ad43fb7b3e
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: Acer
	Product Name: Challenger2_SKS
	Version: V1.01
	Serial Number: NBQ171100163704CA27200
	Asset Tag: Type2 - Board Asset Tag
	Features:
```

---------------------------------------
sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.aml - doesn't return anything.

I tried manually going into the DSDT folder but it won't allow me to do so.

kroaton, 2021-10-10:
[  141.540666] acer_wmi: Unknown function number - 7 - 2
[  141.732746] acer_wmi: Unknown function number - 7 - 2

JafarAkhondali, 2021-10-14:
Change this line:
https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/c0f99438d7f684d12b03ff811705ca80f7ecf51a/src/facer.c#L2549

to:
`			if (return_value.key_num == 0x2)`

and add this:
```
{
				.callback = dmi_matched,
				.ident = "Acer Predator G9-793",
				.matches = {
						DMI_MATCH(DMI_SYS_VENDOR, "Acer"),
						DMI_MATCH(DMI_PRODUCT_NAME, "Predator G9-793"),
				},
				.driver_data = &quirk_acer_predator_pt917_71,
		},
```
after line 770 in facer.c.

Do a reboot, then try installing.

I have no idea if it would work, I just used similar code for your laptop. Use at your own risk.

kroaton, 2021-10-14:
I'll give it a shot when I get back in town. Cheers!