JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #24: Strange bug on ph315-53

felrugo, 2021-11-15:
I found a strange bug on my ph315-53.
After installing the module on Manjaro (KDE) when I press the function button to lower the brightness of the keyboard it also zeroes
out the brightness level of the monitor. I need to manually change ot back with the function key.

The strange thing is that this error occurs after every reboot or reload from sleep, but for only once. If it once lower the brightness
of the monitor the bug only occurs after the next reboot.

It happens even after the facer module was unloaded.
The only thing that helped was booting into the BIOS.
After that the bug disappeared until i installed the module again.

JafarAkhondali, 2021-11-17:
Hi
Thanks for reporting the bug. I also experienced this but actually have no clue what is the cause of it 😵‍💫

vivekkj123, 2022-01-28:
This bug is also reproducible in Acer nitro 5 (AN515-56)

JafarAkhondali, 2023-10-13:
@jyash8 I'm not sure but maybe this: https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/a199d77c61e25e7c54b8d31bd0e9398a2ecd6e0e/src/facer.c#L1209