JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #25: Acer Nitro 5 (AN515-57) wmi/devices directory exists but with a prefix

rithvikvibhu, 2021-11-20:
The README says that if this directory exists, it may work. But I seem to have a directory with a prefix `PNP0C14:00-`.

```bash
❯ file /sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/
/sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/: cannot open `/sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/` (No such file or directory)

❯ file /sys/bus/wmi/devices/PNP0C14:00-7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/
/sys/bus/wmi/devices/PNP0C14:00-7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/: directory

# without trailing slash, if it helps
❯ file /sys/bus/wmi/devices/PNP0C14:00-7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56
/sys/bus/wmi/devices/PNP0C14:00-7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56: symbolic link to ../../../devices/pci0000:00/PNP0C14:00/wmi_bus/wmi_bus-PNP0C14:00/PNP0C14:00-7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56

```

device info (Manjaro KDE):
```
❯ sudo dmidecode | grep "Product Name" -B 2 -A 4
System Information
        Manufacturer: Acer
        Product Name: Nitro AN515-57
        Version: V1.10
        Serial Number: **********
        UUID: 680e****-****-****-****-***********
        Wake-up Type: Power Switch
--
Base Board Information
        Manufacturer: TGL
        Product Name: Scala_TLS
        Version: V1.10
        Serial Number: **********
        Asset Tag: Type2 - Board Asset Tag
        Features:
```

Will this work (with or without changes to the code)? Thanks!

JafarAkhondali, 2021-11-21:
Hi,
regardless of the existence of file, I don't think module would work for you.
Current implementations for Nitro series didn't work with turbo mode, and your keyboard is not a 4-zone RGB. If your keyboard supports dynamic effect, it may work but again I'm not sure. I think it should be safe to test.

rithvikvibhu, 2021-11-22:
They keyboard has 4 zones, I'll change the directory string and try it out and keep this issue updated.

rithvikvibhu, 2021-11-22:
Install:

```
❯ sudo ./install.sh
make -C /lib/modules/5.15.2-2-MANJARO/build M=/home/rithvik/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/5.15.2-2-MANJARO/build'
  CC [M]  /home/rithvik/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
  MODPOST /home/rithvik/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/Module.symvers
  CC [M]  /home/rithvik/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.mod.o
  LD [M]  /home/rithvik/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
  BTF [M] /home/rithvik/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
make[1]: Leaving directory '/usr/lib/modules/5.15.2-2-MANJARO/build'
rmmod: ERROR: Module acer_wmi is not currently loaded
[49741.148579] audit: type=1130 audit(1637558518.634:217): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=packagekit comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[49741.229113] audit: type=1106 audit(1637558518.714:218): pid=51010 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/3 res=success'
[49741.229125] audit: type=1104 audit(1637558518.714:219): pid=51010 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/3 res=success'
[50023.689786] audit: type=1100 audit(1637558801.176:220): pid=53478 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:authentication grantors=pam_faillock,pam_permit,pam_faillock acct="rithvik" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[50023.690839] audit: type=1101 audit(1637558801.176:221): pid=53478 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="rithvik" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[50023.690976] audit: type=1110 audit(1637558801.176:222): pid=53478 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[50023.692078] audit: type=1105 audit(1637558801.179:223): pid=53478 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[50025.000191] facer: Acer Laptop ACPI-WMI Extras
[50025.000218] facer: Function bitmap for Communication Button: 0x801
[50025.006829] input: Acer WMI hotkeys as /devices/virtual/input/input34
[*] Done
```

Right after the install and every time the python script is run, the keyboard blinks off for a short while (~0.2 secs). But the colors don't change.

So guess it's not possible even if the keyboard has 4 zones? or is there any hope :p

rithvikvibhu, 2023-03-26:
Randomly decided to check back and came across https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/21. Tried out the latest code and it works! Thanks @hornc

closed 2023-03-26 by rithvikvibhu