JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #26: Predator PT515-52

ayan-r, 2021-11-30:
Does not work for my laptop.
will there be updates for this model? 

System Information
        Manufacturer: Acer
        Product Name: Predator PT515-52
        Version: V1.07
        Serial Number: NHQ6WER004030039896600
        UUID: d1c70cab-cd72-11ea-b7bd-3f18c8ac802e
        Wake-up Type: Power Switch**

Base Board Information
        Manufacturer: CML
        Product Name: Spyder_CMS
        Version: V1.07
        Serial Number: NBQ6W11002030001066600
        Asset Tag:                       
        Features:

file /sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/
/sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/: cannot open `/sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/' (No such file or directory)

[dsdt.aml.txt](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/7625380/dsdt.aml.txt)

JafarAkhondali, 2021-12-03:
Hi,
Looks like your model uses a different approach to handle gaming functions from known models (like mine). Sorry but currently I can't work on it as I don't have enough resources for working on this feature, and it's hard to maintain open source projects.
If you are interested, I gave some details for developing the feature yourself here https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/21

closed 2021-12-06 by ayan-r

ayan-r, 2021-12-06:
Thanks for the answer, I will try to do it