JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #27: Predator PH315-54 - Turbo Mode support

SuperSecureHuman, 2021-12-03:
Hi, 
The keyboard light works perfectly, but the turbo button dosen't

My system Specs
```
sudo dmidecode -s system-product-name
Predator PH315-54
```
```
OS: Arch Linux x86_64
Kernel: 5.15.5-zen1-1-zen
```

Actually, I am not sure what info this issue/support needs, letme know if you need more info.

Again, all the keyboard light effects work, but not turbo button

Also, does the predator sense/nitro sense button can be used on Linux for something? Does acer wmi give support for that ?

Thanks!

JafarAkhondali, 2021-12-03:
Hi,  
Can you tell me the count of your fans per CPU and GPU?  
I can try adding your model with knowing fans count, but not sure if it works.  
I think it should be possible to use the predatorSense key, but I have no idea how to use it, please tell me if you found a way for it.

SuperSecureHuman, 2021-12-03:
By physically seeing, I see one fan for cpu, and one for gpu.

Lemme know if looks are deceiving 🧐

And I am not removeing the back cover, because warranty.

And sure, I will look into predator sense key.
If I get more time, I would be happy to combine this project along with nbfc-linux, and make an app for Linux 😋

JafarAkhondali, 2021-12-03:
Ok I've added the required code for your model. Uninstall the module, reboot, clone the new version, and install again. I hope it would work for you

SuperSecureHuman, 2021-12-04:
Turbo mode is able to ramp up my fans!
But I don't see the overclock being applied.

I tried doing a cuda accelerated AI model training, and monitored the GPU via nvtop. The cuda clocks were around 1990MHz, and memory was at 7000MHz.

I need some time to check with windows, to see if any GPU overclock is being applied in the first place!

Again, the button works, and my fans go to max speed, need to confirm about overclock by going into windows.

Thanks!

SuperSecureHuman, 2021-12-04:
There is a 100MHz overclock on both GPU cores and Memory Cores on windows side, but not on linux side. And I dont think that is gonna do a huge performance difference. So, I am happy with my current status of this button working to ramp up the fans.

If you wish to try get that 100Mhz overclock working on my laptop, you may suggest me ideas, else you can close this issue.

I will reopen this issue, if I find a way to use that predator Sense button on linux

Thank you for this work!

JafarAkhondali, 2021-12-04:
I've tried implementing exactly what happened on predator sense for turbo mode, on Linux. And validated the result using http://www.geeks3d.com. It gave better performance on turbo mode for me, can you try this as well?

SuperSecureHuman, 2021-12-06:
I am now getting this error, lemme fix it and let you know

```
[ 6062.793714] facer: version magic '5.15.5-zen1-1-zen SMP preempt mod_unload ' should be '5.15.6-zen2-1-zen SMP preempt mod_unload '
```

SuperSecureHuman, 2021-12-09:
Couldnt get benchmark mode on Geeks 3d working properly
So I tested with unigene

Here are the results

Just fan at 100%
[image]

Turbo Mode ON
[image]

During Testing, I noticed that GPU clocks never touched 2000+
So I locked GPU clocks at 2000 via nvidia-smi (but still during benchmark, it clocked down)

Here are those results
Clocked fan at 100%
[image]

Clocked Turbo Mode
[image]

All of these results seem to be in margin of error

JafarAkhondali, 2021-12-09:
@SuperSecureHuman All these tests are using Intel Graphics GPU with i915 driver. I think GPU is not involved in these at all.

SuperSecureHuman, 2021-12-09:
I used the prime render offload to offload it to GPU, and I had GPU monitored for the whole.
I did see the GPU being used

As for why unigene reports like that, I am not sure about it.

Plus, I don't think that amount score is possible with igpu at 4k 🤔