JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #28: Any possible support for Secure Boot

Aakarshit-Sharma19, 2021-12-11:
Hey,
Is there any way to enable secure boot for the driver and if possible using changes, what should be the changes done to the module.
Thanks

JafarAkhondali, 2021-12-11:
No
I don't think it would be possible, as secure boot is trying to disable loading 3rd party modules somehow

mirh, 2021-12-21:
It's not up to single modules to sign themselves. 
Upstreaming would handle this, but I don't think it's a priority atm.

AP-XD, 2022-02-28:
Is it possible to sign the module which its trying to load?
Since disabling secure boot is an issue with WIN 11 since it needs secure boot so I can't dual boot :(

Aakarshit-Sharma19, 2022-02-28:
@AP-XD That's great. Can you share the steps?

mirh, 2022-02-28:
W11 requires secure boot support. It doesn't require you to keep it enabled.

AP-XD, 2022-02-28:
> W11 requires secure boot support. It doesn't require you to keep it enabled.

Yeah actually yes but in my bios all options were locked out
Until today when I came to know that setting supervisor password unlocks them so I signed the module

AP-XD, 2022-02-28:
> @AP-XD That's great. Can you share the steps?

Yeah I made a guide for my own future reference I would share that here 

`git clone https://github.com/JafarAkhondali/acer-helios-300-rgb-keyboard-linux-module
cd "acer-helios-300-rgb-keyboard-linux-module"
chmod +x ./*.sh
sudo ./install_service.sh`

Now We need to sign the module

`cd src
sudo openssl \
    req \
    -new \
    -x509 \
    -newkey \
    rsa:2048 \
    -keyout MOK.priv \
    -outform DER \
    -out MOK.der \
    -days 36500`

enter all var it asks for like  organization password and all

`sudo chmod 600 MOK*
sudo mokutil --import MOK.der`

reboot and enroll MOK in BIOS

`sudo KBUILD_SIGN_PIN="your password which u used to create the key" /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 ./MOK.priv ./MOK.der facer.ko
sudo insmod facer.ko
modinfo facer.ko
cd ..`

WAVE WHICH I LIKE THE MOST

`sudo ./facer_rgb.py -m 3 -s 5 -b 100`

[Reference 1](https://gist.github.com/reillysiemens/ac6bea1e6c7684d62f544bd79b2182a4)
[Reference 2](https://askubuntu.com/questions/840086/ubuntu-16-10-virtualbox-doesnt-get-signed)

PR #96 (2023-05-16): Add secure boot support for Ubuntu
Partly resolve #28 

Sign the kernel module in the makefile for secure boot in Ubuntu: https://wiki.ubuntu.com/UEFI/SecureBoot/Signing
closed 2023-05-16 by JafarAkhondali