JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #31: Not working in my helios 300 2021

Shawshank-dulal, 2022-02-11:
Output of "sudo dmidecode | grep "Product Name" -B 2 -A 4"
System Information
	Manufacturer: Acer
	Product Name: Predator PH315-54
	Version: V1.05
	Serial Number: NHQC2SG009132148B83400
	UUID: bcb371cd-f1e9-eb11-80e1-088fc30c3654
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: TGL
	Product Name: QX60_TLS
	Version: V1.05
	Serial Number: NBQC21100213292BEA3400
	Asset Tag: Type2 - Board Asset Tag
	Features:
dsdt output: [https://we.tl/t-tkSr57I0k6](url)

JafarAkhondali, 2022-02-13:
Any errors?

Shawshank-dulal, 2022-02-16:
Turbo led lights up after pressing turbo key but the rgb functionality does not work and also does not produce any errors

JafarAkhondali, 2023-10-07:
Can you check the issue again with latest version?  Uninstall - reboot - reinstall