JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #32: Unexpected behavior on Predator PH315-54

egarcia1360, 2022-02-11:
`System Information
	Manufacturer: Acer
	Product Name: Predator PH315-54
	Version: V1.02
	Serial Number: *
	UUID: *
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: TGL
	Product Name: QX60_TLS
	Version: V1.02
	Serial Number: *
	Asset Tag: Type2 - Board Asset Tag
	Features:`

I was able to install without any issues, but then the following occurred:

The only lighting option that I found to work properly was changing the backlight brightness, which I was already able to do with the Fn keys. I tried a few different mode and color options, none of which seemed to do anything. Interestingly, when setting the brightness to anything other than the defaults (0, 25, 50, 75, 100), pressing the "Dim Backlight" key performs its intended function a couple of times before beginning to cycle through various shades and brightness levels of blue and green, ultimately going back to normal after a while, depending on the brightness value used. [Here](https://imgur.com/a/PsnTFSw) is a short video to demonstrate how it reacts to `-b 42`, for example.

I'm not sure about the Turbo button, as it lit up but I didn't hear any changes in fan noise; I'll have to try running some games or benchmarks later on and get back to you.

JafarAkhondali, 2022-02-13:
I received several issues for PH315-54 model, so I assume there are some differences for this model. Unfortunately I can't put time on de-compiling code for this model in near future

snowyoneill, 2022-03-14:
Just an FYI, I have the PH315-54-760S and your kernel module worked perfectly on my machine - after a bit of back and forth trying to get it installed.

JafarAkhondali, 2022-03-20:
@snowyoneill thanks, do you mind sharing what changes have you made?

snowyoneill, 2022-05-08:
I had to follow the instructions for the install failure on the PH513-35:
https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/4

I added these 2 lines to the bottom of the header section of facer.c:

```
#include <linux/uaccess.h>
#include <asm/uaccess.h>
```

Once i did that i managed to get the kernel driver installed and everything works as expected.