JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #35: Predator PH317-55 issue

yaojiunn, 2022-02-20:
Dear Sir,
I am trying to use this module on PH317-55. After the procedure of Install one time (Module won't work after reboot),
[image]
Turbo mode & RGB doensn't work. Did I miss some procudure?

mmsaeed509, 2022-02-20:
Hi @yaojiunn, there are two ways to install 
the first one is
[Install one time](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module#:~:text=Install%20one%20time%20(Module%20won%27t%20work%20after%20reboot)) Module won't work after reboot

and the second one is 
[Install as a service](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module#:~:text=Install%20as%20a%20service%20(Will%20work%20after%20reboot))  her you install the module as system services(systemd) ,  the module will work after reboot

JafarAkhondali, 2022-02-20:
@yaojiunn you need to disable secure boot in BIOS. But your model is not supported ( at least I think the turbo button won't work). If you have a 4-zone LED keyboard, the keyboard may work.