JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #37: Acer Predator Helios 300 Issue (Predator PH315-54)

Setu-Gupta, 2022-03-08:
I'm running arch linux. When I load a profile using the facer_rgb.py script, nothing happens. Moreover my keyboard backlight is set to light blue color by default. This does not change after running the script.

`uname -a` gives:
`Linux acerarch 5.16.12-arch1-1 #1 SMP PREEMPT Wed, 02 Mar 2022 12:22:51 +0000 x86_64 GNU/Linux`

`sudo dmidecode | grep "Product Name" -B 2 -A 4` gives:
`System Information
	Manufacturer: Acer
	Product Name: Predator PH315-54
	Version: V1.11
	Serial Number: NHQC1SI00514725E7A3400
	UUID: 36f47361-b94d-ec11-80e2-088fc32ea456
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: TGL
	Product Name: QX60_TLS
	Version: V1.11
	Serial Number: NBQC11100214771B483400
	Asset Tag: Type2 - Board Asset Tag
	Features:
`

The dsdt.aml file is here: https://drive.google.com/file/d/15y4UxmqmL1W8-u7SaZeyYb9DDLysGtvV/view?usp=sharing

JafarAkhondali, 2022-03-20:
Did you receive any errors on installation?

Setu-Gupta, 2022-03-21:
No I didn't face any issues on installation

morais948, 2022-08-22:
Também estou com o mesmo problema, ocorre tudo bem na instalação mas nada acontece com nenhum comando. 
Uso o kali linux

JafarAkhondali, 2022-10-02:
Check again to see if it's fixed. Please test both turbo mode and RGBs on different OSs.

Setu-Gupta, 2022-10-02:
It is not working. Again the installation threw no error. When I tried running `./facer_rgb.py -m 1 -s 4 -b 100 -cR 255 -cG 0 -cB 255`, there was no change to my keyboard backlight. Also when I tried pressing the Turbo button, there was no change, whereas pressing the turbo button on windows ramps up the fan speed.

mtorsiello, 2022-11-05:
See here https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/53#issuecomment-1304608901 for my comment on how I got thus working on PH315-55. It's a bit of a chore to do it at first, as I manually edit the facer source file to include a few different changes from several threads into one.