JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #38: Failed install PH315-52

egefeyzioglu, 2022-03-11:
Ubuntu 20.04.4 LTS running on PH315-52

CC is complaining about `copy_from_user` not being declared, even though kernel headers are installed. Here is the complete terminal output
```

egef@ege-pc:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-5.4.0-100-generic is already the newest version (5.4.0-100.113).
The following packages were automatically installed and are no longer required:
  arduino-core avr-libc avrdude binutils-avr extra-xdg-menus gcc-avr libbdplus0 libftdi1 libhidapi-libusb0 libjna-java libjna-jni librxtx-java libusb-0.1-4 linux-headers-5.4.0-99 linux-headers-5.4.0-99-generic
  linux-image-5.4.0-99-generic linux-modules-5.4.0-99-generic linux-modules-extra-5.4.0-99-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 17 not upgraded.

egef@ege-pc:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo ./install.sh
make -C /lib/modules/5.4.0-100-generic/build M=/home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-100-generic'
  CC [M]  /home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘gkbbl_drv_write’:
/home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1948:8: error: implicit declaration of function ‘copy_from_user’ [-Werror=implicit-function-declaration]
 1948 |  err = copy_from_user(config_buf, buf, GAMING_KBBL_CONFIG_LEN);
      |        ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:270: /home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[1]: *** [Makefile:1762: /home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-100-generic'
make: *** [Makefile:9: default] Error 2
rmmod: ERROR: Module acer_wmi is not currently loaded
insmod: ERROR: could not load module src/facer.ko: No such file or directory
[79098.360173] audit: type=1400 audit(1646966565.967:1805807): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=154869 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[79098.360195] audit: type=1400 audit(1646966565.967:1805808): apparmor="DENIED" operation="open" profile="snap.discord.discord" name="/proc/2225/cmdline" pid=154869 comm="Discord" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[79098.360282] audit: type=1400 audit(1646966565.967:1805809): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=154869 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[79098.360289] audit: type=1400 audit(1646966565.967:1805810): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=154869 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[79098.360339] audit: type=1400 audit(1646966565.967:1805811): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=154869 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[79098.360404] audit: type=1400 audit(1646966565.967:1805812): apparmor="DENIED" operation="open" profile="snap.discord.discord" name="/proc/2238/cmdline" pid=154869 comm="Discord" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[79098.360433] audit: type=1400 audit(1646966565.967:1805813): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=154869 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[79098.360487] audit: type=1400 audit(1646966565.967:1805814): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=154869 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[79098.360609] audit: type=1400 audit(1646966565.967:1805815): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=154869 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[79098.360616] audit: type=1400 audit(1646966565.967:1805816): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=154869 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[*] Done
```

JafarAkhondali, 2022-03-11:
Possible duplicate https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/4

egefeyzioglu, 2022-03-12:
Unfortunately not

```
egef@ege-pc:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo bash ./install.sh
make -C /lib/modules/5.4.0-104-generic/build M=/home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-104-generic'
  CC [M]  /home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘gkbbl_drv_write’:
/home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1948:8: error: implicit declaration of function ‘copy_from_user’ [-Werror=implicit-function-declaration]
 1948 |  err = copy_from_user(config_buf, buf, GAMING_KBBL_CONFIG_LEN);
      |        ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:270: /home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[1]: *** [Makefile:1762: /home/egef/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-104-generic'
make: *** [Makefile:9: default] Error 2
insmod: ERROR: could not load module src/facer.ko: No such file or directory
[ 7815.563687] audit: type=1400 audit(1647052077.541:144552): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=10389 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[ 7815.563753] audit: type=1400 audit(1647052077.541:144553): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=10389 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[ 7815.563820] audit: type=1400 audit(1647052077.541:144554): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=10389 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[ 7815.563888] audit: type=1400 audit(1647052077.541:144555): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=10389 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[ 7815.563952] audit: type=1400 audit(1647052077.541:144556): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=10389 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[ 7815.564020] audit: type=1400 audit(1647052077.541:144557): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=10389 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[ 7815.564095] audit: type=1400 audit(1647052077.541:144558): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=10389 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[ 7815.564154] audit: type=1400 audit(1647052077.541:144559): apparmor="DENIED" operation="ptrace" profile="snap.discord.discord" pid=10389 comm="Discord" requested_mask="read" denied_mask="read" peer="unconfined"
[ 7818.021404] acer_wmi: Unknown function number - 11 - 0
[ 7821.547716] acer_wmi: Acer Laptop WMI Extras unloaded
[*] Done
```

(Also tried `apt update`, no new packages found.)

JafarAkhondali, 2022-03-12:
@egefeyzioglu  Did you try adding `#include <asm/uaccess.h>` or `#include <linux/uaccess.h>` in https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/main/src/facer.c ?

egefeyzioglu, 2022-03-16:
Fixed thank you!

closed 2022-03-16 by egefeyzioglu