JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #40: DId not work after restart.

dragonfetus, 2022-03-27:
It worked as expected when I first installed it but after shutting down my computer and turning it on after a few hours the colors were gone and the keyboard was stuck at red. I did install it as a service so that should not have happened. now if I try to change the colours using the python script this error appears. `./facer_rgb.py -m 3 -s 5 -b 100             ──(Sun,Mar27)─┘
Traceback (most recent call last):
  File "/home/dragonfetus/acer-predator-turbo-and-rgb-keyboard-linux-module/./facer_rgb.py", line 203, in <module>
    with open(CHARACTER_DEVICE, 'wb') as cd:
PermissionError: [Errno 13] Permission denied: '/dev/acer-gkbbl-0'`

I tried reinstalling but it did not help.

dragonfetus, 2022-03-27:
I tried rebooting my laptop along with other things and the script works now. But after restart the colors go back to red and I have to run the script every time I restart.

JafarAkhondali, 2022-03-27:
there are two install modes, if you want to make script after reboot, try installing it [as service(systemd perhaps)](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module#install-as-a-systemd-service-will-work-after-reboot)

dragonfetus, 2022-03-28:
I have installed it as a systemd service and as an openrc service , none of them works.

dragonfetus, 2022-03-28:
the service created by the script is active (turbo-fan.service) but the colors do not stay permanent. The colors are fine when I reboot but go back to red when I completely power off and on my system.

dragonfetus, 2022-04-01:
I have fixed the issue by setting a cronjob to change the lights as a temporary fix, the service still does  not work.

closed 2022-04-01 by dragonfetus