JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #41: Acer Nitro-5 2021 Issue

shakyasaijal, 2022-04-05:
Device Info

`$ sudo dmidecode | grep "Product Name" -B 2 -A 4`

````
System Information
	Manufacturer: Acer
	Product Name: Nitro AN515-45
	Version: V1.08
	Serial Number: ************************************
	UUID: *****-****-****-****-**********
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: CZ
	Product Name: Scala_CAS
	Version: V1.08
	Serial Number: ************************
	Asset Tag: Type2 - Board Asset Tag
	Features:
````

When  `$ sudo ./install.sh` command was used, following error occurred:
[image]

OS: Linux 20.04

JafarAkhondali, 2022-04-05:
disable secure boot (easiest way)
or https://unix.stackexchange.com/questions/543576/modprobe-fails-with-operation-not-permitted
or [sign the module](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/28)

jaravskie, 2022-07-31:
Hello there,

Have you tried it installing it as a service?

I also have the same issue sometimes when i installed it using the "Install One Time Method", i happen to repeat the installing via ./install.sh and ./install_service.sh and that just caused error similar to this, i had to remove the directory and try again, also before attempting i have disabled Secure Boot and TPM, and i have the same model and bios version as you, what worked for me is to install it as "Install as a systemd service" and i have to use the ./facer_rgb.py command every startup, i wanted to turn off the rgb during the day, i am using Manjaro and have nbfc installed along side the service, the program works as expected except no turbo mode of course, clicking the nitro button does nothing.