JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #42: PredatorNonSense

kphanipavan, 2022-04-08:
Hello,

*This is not a issue, just a self promotion.*

I made a program called [PredatorNonSense](https://github.com/kphanipavan/PredatorNonSense). This program partially fulfills the purpose of [Predator Sense](https://www.acer.com/ac/en/IN/content/predatorsense) that is available only on Windows.

It supports Fan Mode and Speed Control for both of the fans available on my laptop. To make this work, I had to make EC writable and programmed relevant write operations with an average looking UI made with Qt. I managed to make launching the app with PredatorSense Key work as well as setting the Fan Profile that happens with GPU Boost*. Keyboard lighting partially works and has to be worked upon by myself.

After trying to make the turbo button work for a very long time, I just stumbled upon this repo and was surprised how easy it was to make it work with a modified kernel module. I tried incorporating the Turbo Button toggle into my program but, due to me being a Linux beginner and my lack of familiarity with C, I understood nothing and hence made this post.

I was hoping that linking my project here can help with the development of this project in the field of Fan Control and other possible aspects.

Please give [PNS](https://github.com/kphanipavan/PredatorNonSense) a try if possible. This also helps me with the development of PNS.

Thank you.

 

*These 2 features are yet to be pushed to master.

NikitaGolovaniuk, 2022-04-08:
Will you wide model list? I need it for my acer predator triton 300 se. Or just any program to control fan/rgb/turbo

kphanipavan, 2022-04-08:
> Will you wide model list? I need it for my acer predator triton 300 se. Or just any program to control fan/rgb/turbo

Providing various model support is a planned feature...
Though it requires community support.
You will have to manually check the EC compatibility and provide 
Open an issue over there if you wish to track progress...

NikitaGolovaniuk, 2022-04-15:
> > Will you wide model list? I need it for my acer predator triton 300 se. Or just any program to control fan/rgb/turbo
> 
> Providing various model support is a planned feature... Though it requires community support. You will have to manually check the EC compatibility and submit it. Current program doesn't require much refactoring for variable model support... Open an issue over there if you wish to track progress...

can you explain how to make turbo button work? I've found how to make it light or how to control everything by AC codes(turbo mode too) but i cant find keycode for turbo button. It looks like it isnt a part of keyboard idk

JafarAkhondali, 2022-04-17:
> can you explain how to make turbo button work? I've found how to make it light or how to control everything by AC codes(turbo mode too) but i cant find keycode for turbo button. It looks like it [...]

Here is what I did:
1. [Log unknown key number events](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/e84ab34e0eea023bc320ac71a522f559958d9bb3/src/facer.c#L2535) (the log will be available in journalctl). Using this, you can determine turbo button key event and trigger your custom functions for the event
2. Detect state of turbo mode, by querying WMI to find if the LED below turbo button is on or off.
3. Toggle fan speed to turbo or normal, based on current turbo mode state. This requires calculating a custom payload based on count of CPU and GPU, and call the required WMI function. Here is the I've ported from official Acer PredatorSense https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/e84ab34e0eea023bc320ac71a522f559958d9bb3/src/facer.c#L1770
4. Toggle overclock or normal speed for CPU and GPU (again using WMI). You can find the code her https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/e84ab34e0eea023bc320ac71a522f559958d9bb3/src/facer.c#L2234

bnnthang, 2022-06-12:
@kphanipavan Hey nice work. I wonder if I press the Predator key will there be a GUI show up? I'm trying to make the same thing for laptop - hope I can learning something from your repo

kphanipavan, 2022-06-12:
@bnnthang Thank you...

I use a tool called [Evtest](https://manpages.ubuntu.com/manpages/trusty/man1/evtest.1.html). This puts all the events from a device to STDOUT as below.

[image]
 
I then find the event that occurs when I press the PredatorSense Key, which, in my case, is shown as highlighted above. I then grep the output until it occurs once, using the the -m flag, and I put my program to be executed at the end when the key is pressed.

My use case: [Recursively calling while waiting till the special key is pressed](https://github.com/kphanipavan/PredatorNonSense/blob/master/PNS.sh#L6)

[image]

Above is the flow which works for me...

kphanipavan, 2022-06-12:
Am closing this...

closed 2022-06-12 by kphanipavan