JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #44: Not working on PH315-54

Ameen-Sha-Cheerangan, 2022-04-13:


Ameen-Sha-Cheerangan, 2022-04-13:
```
ameen@pop-os:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo dmidecode | grep "Product Name" -B 2 -A 4
System Information
	Manufacturer: Acer
	Product Name: Predator PH315-54
	Version: V1.07
	Serial Number: NHQC5SI00613302AB13400
	UUID: 794cb464-dff8-eb11-80e1-088fc310b710
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: TGL
	Product Name: QX60_TLS
	Version: V1.07
	Serial Number: NBQC511002133A65BC3400
	Asset Tag: Type2 - Board Asset Tag
	Features:
ameen@pop-os:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.aml
ameen@pop-os:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ 

```

snowyoneill, 2022-05-08:
@Ameen-Sha-Cheerangan check out my answer on https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/32

See if that helps - i did manage to get the kernel module installed on my PH514-54.

closed 2022-05-09 by Ameen-Sha-Cheerangan