JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #47: RGB timeout changed!

dhruv8sh, 2022-05-19:
Model : Nitro AN515-56
 Script started working fine recently until I noticed that the RGB timeout was very little (about 3 secs).

JafarAkhondali, 2022-05-19:
Wait ... what? what do you mean by timeout :)) 
I have no idea why it didn't work, and why it's working now

dhruv8sh, 2022-05-19:
So I had [https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/43](this issue) open because it wasn't working on my model. Turns out, it works fine on EndeavourOS and not on Manjaro(both with default packages). Probably, something related to packages.

By a 3 sec timeout, I mean that when I am inactive/away from the keyboard, the backlight only stays on for 3 secs.

Thanks for this amazing project, btw!

JafarAkhondali, 2022-05-19:
@dhruv8sh I think you can change this setting in bios.

dhruv8sh, 2022-05-19:
Yes @JafarAkhondali , I am aware of the setting to turn off/on the backlight timeout, but the problem is I want it on but I don't want it too short...  Is there a variable I can edit to change the timeout?

I, however, will change this from windows for now.