JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #49: Not working on fedora 36

ramonssouza, 2022-06-07:
PH315-52

[ramonss@fedora acer-predator-turbo-and-rgb-keyboard-linux-module]$ systemctl status turbo-fan.service
● turbo-fan.service - Enables turbo button
     Loaded: loaded (/etc/systemd/system/turbo-fan.service; enabled; vendor preset: disabled)
     Active: active (exited) since Tue 2022-06-07 10:09:37 -03; 1min 39s ago
    Process: 31745 ExecStart=/bin/bash ./install.sh (code=exited, status=0/SUCCESS)
   Main PID: 31745 (code=exited, status=0/SUCCESS)
        CPU: 1.194s

jun 07 10:09:38 fedora bash[32086]: [19109.061207] systemd-sysv-generator[30928]: SysV service '/etc/rc.d/init.d/livesys' lacks a native systemd unit file. Automatically generating a unit file for compatibility. Please update package to >
jun 07 10:09:38 fedora bash[32086]: [19109.061229] systemd-sysv-generator[30928]: SysV service '/etc/rc.d/init.d/livesys-late' lacks a native systemd unit file. Automatically generating a unit file for compatibility. Please update packag>
jun 07 10:09:38 fedora bash[32086]: [19109.287962] systemd-sysv-generator[31013]: SysV service '/etc/rc.d/init.d/livesys' lacks a native systemd unit file. Automatically generating a unit file for compatibility. Please update package to >
jun 07 10:09:38 fedora bash[32086]: [19109.287989] systemd-sysv-generator[31013]: SysV service '/etc/rc.d/init.d/livesys-late' lacks a native systemd unit file. Automatically generating a unit file for compatibility. Please update packag>
jun 07 10:09:38 fedora bash[32086]: [19110.407002] acer_wmi: Acer Laptop WMI Extras unloaded
jun 07 10:09:38 fedora bash[32086]: [19241.137849] acer_wmi: Acer Laptop ACPI-WMI Extras
jun 07 10:09:38 fedora bash[32086]: [19241.137868] acer_wmi: Function bitmap for Communication Button: 0x800
jun 07 10:09:38 fedora bash[32086]: [19241.148368] input: Acer WMI hotkeys as /devices/virtual/input/input38
jun 07 10:09:38 fedora bash[32086]: [19252.322371] acer_wmi: Acer Laptop WMI Extras unloaded
jun 07 10:09:38 fedora bash[31745]: [*] Done

mmsaeed509, 2022-06-07:
@ramonssouza Install as a systemd service, it will work after restarting the machine
```bash
chmod +x install_service.sh
sudo ./install_service.sh
```

ramonssouza, 2022-06-07:
@mmsaeed509 still not working, I installed it this way, but it doesn't work.

ramonssouza, 2022-07-11:
SELinux:

type=AVC msg=audit(1657512697.569:539): avc:  denied  { module_load } for  pid=5596 comm="insmod" path="/opt/turbo-fan/src/facer.ko" dev="nvme0n1p3" ino=361064 scontext=system_u:system_r:kmod_t:s0 tcontext=system_u:object_r:usr_t:s0 tclass=system permissive=0

Solution:
semanage fcontext -a -t modules_object_t /opt/turbo-fan/src/facer.ko

closed 2022-07-11 by ramonssouza