JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #52: PH315-52 Kernel panic

ramonssouza, 2022-07-09:
What Linux distribution do you use? I can't use it on Manjaro and Fedora 36

ramonssouza, 2022-07-09:
[image]

ramonssouza, 2022-07-09:
[image]

ramonssouza, 2022-07-14:
opensuse tumbleweed working perfectly fine

JafarAkhondali, 2022-07-20:
@ramonssouza I'm using Pop-os as it works out of the box with Nvidia graphic cards.

ramonssouza, 2022-08-01:
Happens with pop-os 22.04 too, when I turn off the notebook this problem appears

wagyourtail, 2022-08-01:
I've had this as well on Arch (I also have a ph315-52) didn't realize it was caused by this...
doesn't really inconvenience me too much, I just hard power off when it comes up because it only happens when I'm shutting down the system anyway.

JafarAkhondali, 2022-08-01:
I can't debug it as I don't own my Acer Predator anymore(physical damage). Hope someone else can work on this

ramonssouza, 2022-08-03:
I tested on artix linux with openrc and it seems to work without problems

ramonssouza, 2022-08-03:
Will agum error with systemd service?

ramonssouza, 2022-10-14:
Is there any news regarding this bug?

x211321, 2022-10-22:
I’m running the module on a Nitro AN515-45 (4 RGB zones / no turbo). The module worked well with my previous Linux Mint setup (besides the static RGB mode not working). 

A couple of days ago I switched to fedora 36 and now I’m experiencing what I assume to be this exact issue: sporadic kernel panic during shutdown (though on the bright side the static mode is now working).

uname -r
`5.19.16-200.fc36.x86_64`

Since I know nothing about the development and debugging of kernel modules I poked around the script files and started to wonder whether the unloading of the `facer` module during `ExecStop` and the subsequent reload of the `acer_wmi` module could be the cause of this problem.

So I removed the `ExecStop` key from the service configuration and now the issue seems to be gone. Before the change I had kernel panics roughly every second or third reboot. Now after roughly 20 reboots not a single panic occurred. 

Of course this could just be coincidence but maybe its a clue to what is happening. Maybe someone else could try those steps and check if the behavior is reproducible for them.

**Steps I took:**
1) check out the repo
2) If the module/service is already installed run: `uninstall_service.sh`
3) Remove the line `ExecStop=/bin/bash ./uninstall.sh` from  `install_service.sh`
4) Run `install_service.sh`
5) If SELinux protection is active (fedora default) run:

`semanage fcontext -a -t object_r /opt/turbo-fan/src/facer.ko`
`chcon -t modules_object_t /opt/turbo-fan/src/facer.ko`

Otherwise the install script run by systemd cannot access the module.

**Alternatively if everything is already set up and working:**
Removing the `ExecStop` key from `/etc/systemd/system/turbo-fan.service` should have the same effect.

Of course once those modifications have been made `systemctl stop turbo-fan.service` will no longer have any effect. To restore the stock `acer_wmi` module you could either run `/opt/turbo-fan/uninstall.sh` or `rmmod facer` and `modprobe acer_wmi` manually.

Again I have no idea if this is actually the cause of this issue but to me it appears as if the kernel is simply unhappy to have those two modules switched during shutdown.

JafarAkhondali, 2022-10-23:
Can someone verify @x211321 solution?

ramonssouza, 2022-10-23:
I'll test it on my notebook, thanks in advance for the help

ramonssouza, 2022-11-08:
It seems to work fine now. Thanks!

pratyush3757, 2023-01-11:
The fix mentioned by @x211321 works for me too. I tried it on PH315-52, Manjaro, one month ago. And didn't have any panics during shutdown/reboot since.

commit 1c1a0e5c (2023-01-12): Fixk kernel panic

Fixes #52 by removing exec_stop
closed 2023-01-12 by JafarAkhondali