JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #53: Doesn't work on Acer Predator PH315-55, for now.

raystriker, 2022-07-22:
Made the following changes to /src/facer.c to add support for my 2022 Predator PH315-55:
```
static struct quirk_entry quirk_acer_predator_ph315_55 = {
		.turbo = 1,
		.cpu_fans = 1,
		.gpu_fans = 1,
};

```
```

{
	.callback = dmi_matched,
	.ident = "Acer Predator PH315-55",
	.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "Acer"),
			DMI_MATCH(DMI_PRODUCT_NAME, "Predator PH315-55"),
	},
	.driver_data = &quirk_acer_predator_ph315_55,
},
```
Was able to install this as a systemd service without any issues. Unfortunately the turbo mode fan does work, and neither does changing the RGB keyboard profiles. However, the settings I had set in the Predator app on windows are working with linux now.
This does exists: /sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/

I'm using a Fedora based distro called Nobara. Is there any additional info I could provide?

mtorsiello, 2022-11-05:
I'm also using Nobara and this model with no luck. I followed the install instructions, everything seems to be ok but using the script doesn't seem to have any effect. Any updates on this?

mtorsiello, 2022-11-05:
I got this working using changes from this pull request https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/pull/69 as well as adding the 2 includes in the header and the changes in this thread. The issue is the gpu fan gets set to max initially but i used PredatorNonSense to turn it back to auto mode.