JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #62: PT515-52 - RGB doesn't work

Confuze, 2022-08-13:
I installed it using the method for Arch linux.
Turbo works perfectly fine, but rgb doesn't do anything at all. I think this is related to my model, so is there a way to disable the rgb entirely? Because right now the keys just flash different colors randomly.

Thanks for maintaining this project, the turbo feature works well, so I'm very grateful anyways.

```
$ sudo dmidecode | grep "Product Name" -B 2 -A 4
System Information
	Manufacturer: Acer
	Product Name: Predator PT515-52
	Version: V1.10
	Serial Number: NHQ6XEV002019018496600
	UUID: 3bf346b3-912e-11ea-9ef0-4f6f3ed53850
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: CML
	Product Name: Spyder_CMS
	Version: V1.10
	Serial Number: NBQ6X11002019000366600
	Asset Tag:                       
	Features:
```

```
$ sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.aml
(doesn't output anything at all)
```

mmsaeed509, 2022-08-13:
[PT515-52](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module#:~:text=No-,PT515%2D52,No,-PT917%2D71) model not tested

module files located in `/opt/turbo-fan/`, there is a python script `facer_rgb.py` that controls RGB
```
facer_rgb.py -m 0 -z 1 -cR 4 -cB 181 -cG 172 -b 100
facer_rgb.py -m 0 -z 2 -cR 4 -cB 181 -cG 172 -b 100
facer_rgb.py -m 0 -z 3 -cR 4 -cB 181 -cG 172 -b 100
facer_rgb.py -m 0 -z 4 -cR 4 -cB 181 -cG 172 -b 100
```
set RGB to the default colors

Confuze, 2022-08-16:
> [PT515-52](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module#:~:text=No-,PT515%2D52,No,-PT917%2D71) model not tested
>
> module files located in `/opt/turbo-fan/`, [...]

Yeah, but as I mentioned before the script doesn't do anything