JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #65: Predator PH315-54

krishnan-techconative, 2022-09-16:
Hi, 

Can you please enable this feature of this model as well Acer Predator PH315-54, I see it is not in compatible model list also it didn't work and there was no error while installing or running /facer_rgb.py -m 3 -s 5 -b 100

krishnan-techconative, 2022-09-19:
> @krishnan-techconative i have the same model and its working for me. Make sure to log into Windows and play with Predator Sense in order to reset the registers. Also see #32

I tried adding both the header and when run sudo ./install.sh getting the below error.  Please let me know what am I missing

```
➜  acer-predator-turbo-and-rgb-keyboard-linux-module git:(main) ✗ sudo ./install.sh
make -C /lib/modules/5.15.0-47-generic/build M=/home/krishnan/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-47-generic'
  CC [M]  /home/krishnan/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
  MODPOST /home/krishnan/acer-predator-turbo-and-rgb-keyboard-linux-module/Module.symvers
  CC [M]  /home/krishnan/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.mod.o
  LD [M]  /home/krishnan/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
  BTF [M] /home/krishnan/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
Skipping BTF generation for /home/krishnan/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-47-generic'
insmod: ERROR: could not insert module src/facer.ko: Operation not permitted
[29123.639944] [drm:nv_drm_master_set [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to grab modeset ownership
[29123.640049] [drm:nv_drm_master_set [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to grab modeset ownership
[29123.640112] [drm:nv_drm_master_set [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to grab modeset ownership
[29123.668723] [drm:nv_drm_master_set [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to grab modeset ownership
[29123.668823] [drm:nv_drm_master_set [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to grab modeset ownership
[29123.668881] [drm:nv_drm_master_set [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to grab modeset ownership
[29133.187033] audit: type=1326 audit(1663581123.190:193): auid=1000 uid=1000 gid=1000 ses=24 subj=? pid=827633 comm="pool-org.gnome." exe="/snap/snap-store/592/usr/bin/snap-store" sig=0 arch=c000003e syscall=93 compat=0 ip=0x7f044e58439b code=0x50000
[31313.486487] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[32094.572981] acer_wmi: Acer Laptop WMI Extras unloaded
[32094.623175] Lockdown: insmod: unsigned module loading is restricted; see man kernel_lockdown.7
[*] Done

```

snowyoneill, 2022-09-21:
@krishnan-techconative - try disabling apparmor temporarily: **sudo systemctl disable apparmor**

jayrfs, 2023-01-13:
Hi, It appears that this module is working as intended on PH315-54 according to #73. Hence, I am closing this issue.

Feel free to reopen if you face further problems.

closed 2023-01-13 by jayrfs