JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #66: Turn backlight off after 30 seconds - Predator PH315-52

SayGus, 2022-10-01:
how can i turn off turn backlight off after 30 seconds?

Turbo mode works properly using install_service on Linux Mint 21

System Information:
	Manufacturer: Acer
	Product Name: Predator PH315-52
	Version: V1.12
	Serial Number: NHQ63AL001039268469501
	UUID: d83ab8bf-80d1-4798-b99f-d73849268b4e
	Wake-up Type: Power Switch

mmsaeed509, 2022-10-02:
using predator sense on windows

JafarAkhondali, 2022-10-02:
You can control it in bios settings or in windows, currently it's not implemented in this module

ramonssouza, 2022-10-14:
Is the model working fine using this module?  Mine is having a kernel panic

snowyoneill, 2022-12-27:
@SayGus @mmsaeed509 Now you can use my custom application for controlling this. ![Linux PredatorSense](https://raw.githubusercontent.com/snowyoneill/Linux-PredatorSense/master/LinuxPredatorSense.png
)
Check it out: https://github.com/snowyoneill/Linux-PredatorSense