JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #67: I want on Acer Nitro AN515-54 V1.31

Bibekdhkl, 2022-10-17:


JafarAkhondali, 2022-10-17:
Please provide the following information:  

# Model: `sudo dmidecode | grep "Product Name" -B 2 -A 4`

## Count of keyboard RGB zones: 0

### Count of Cpu Fans:  0
### Count of Gpu Fans: 0  

### RGB keyboard works?  Yes\No
### Turbo button turn on fans?  Yes\No
### Turbo button turn on LED? Yes\No
### Turbo button activates overclock? Yes\No

Bibekdhkl, 2022-10-17:
This is terminal output of the command:-
```
System Information
	Manufacturer: Acer
	Product Name: Nitro AN515-54
	Version: V1.31
	Serial Number: NHQ59EM04Q03902A0E3400
	UUID: f8bdacc7-b1f8-ea11-80d7-089798bf6a30
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: CFL
	Product Name: Octavia_CFS
	Version: V1.31
	Serial Number: NBQ591100503927EFC3400
	Asset Tag: Type2 - Board Serial Number
```
Count of keyboard RGB zones: supports only Red Color and isn't RGB
Count of Cpu Fans: no idea
Count of Gpu Fans: no idea
RGB keyboard works? I don't have RGB keyboard
Turbo button turn on fans? No
Turbo button turn on LED? It doesn't have control over LED
Turbo button activates overclock? No

This is my Laptop : https://www.amazon.com/Acer-i5-9300H-GeForce-Keyboard-AN515-54-5812/dp/B086KJBKDW
I guess from here you can get further details

I also have posted this on Reddit hope you could gather some more info about my laptop: https://www.reddit.com/r/AcerNitro/comments/x8qscs/fancontrol_in_nitro_an51554_on_ubuntu_2204/

JafarAkhondali, 2022-11-30:
Try https://github.com/kphanipavan/PredatorNonSense or https://github.com/hirschmann/nbfc  
Currently I don't have access to my Acer laptop so I can't work on this feature.

snowyoneill, 2022-12-27:
@Bibekdhkl i have something for you :)

![Linux PredatorSense](https://raw.githubusercontent.com/snowyoneill/Linux-PredatorSense/master/LinuxPredatorSense.png
)
Check it out: https://github.com/snowyoneill/Linux-PredatorSense

--

Also I submitted a Acer Helio 300 config to nbfc-linux: https://github.com/nbfc-linux/nbfc-linux/pull/70