JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #68: Acer Nitro 515-58, not working

deppy-v2, 2022-10-28:
# Model: Nitro AN515-58

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1 

### RGB keyboard works?  No
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No

dijia1124, 2023-01-31:
Just tested on my AN515-58 running Arch Linux, using "sudo ./install.sh". Keyboard rgb works.

deppy-v2, 2023-05-29:
I reinstalled on Fedora and it works only until reboot. The service installs fine, but does not work after reboot. Am I missing something here?
[image]

JafarAkhondali, 2023-05-29:
@deppy-v2 Yes, you should install it as a service 
`sudo ./install_service.sh`

deppy-v2, 2023-05-31:
Thanks, I see it installed and running, but shows no affect on lights. It only has affect until reboot. After that it resets to default behavior and color. Any clues?

JafarAkhondali, 2023-09-26:
Fixed in last update

closed 2023-09-26 by JafarAkhondali