JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #70: AN517-41 static coloring bug

JMans15, 2022-11-08:
First, thanks a lot for your work! I'm trying to do this for some time now and more or less abandoned when wireshark didn't detect my keyboard so I'm glad someone more talented did it :')

I'm having an issue with static coloring on an AN517-41 (running Fedora 36), some colors work but when I type `./facer_rgb.py -m 0 -cR 235 -cG 219 -cB 178 -b 100 -z 1` it turns the zone white and not the gruvbox-ish brown I was aiming for. 
The strangest thing is that it did in fact work the first time
Everyting else works perfectly btw

JafarAkhondali, 2022-12-07:
Glad it worked for you

Try lowering R,G,B values, ex:
`./facer_rgb.py -m 0 -cR 117 -cG 109 -cB 89 -b 100 -z 1`
or try to play with brightness value.

F-U-B-AR, 2023-03-04:
@JMans15 Hello, would you be able to help me to try and get this running? we have the same specs but I cant figure it out!! 
https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/87

JMans15, 2023-03-04:
@JafarAkhondali Sorry for the late reply, in the end I made my own script to switch between a few modes/colors that I like and didn't think about the issue again, the lower values seem to yield the right colour but the brightness is dimmer than what I can do from windows

@F-U-B-AR I would gladly try to help you fix your problem but I don't think this issue is the right place to do so, my problem has nothing to do with yours, I will comment your issue if I find something

F-U-B-AR, 2023-03-04:
Understood, thank you!