JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #72: Include <linux/uaccess.h> and <asm/uaccess.h>

JafarAkhondali, 2022-11-30:
Some users reported in #32 #44 #53, that it was required to include the mentioned headers in https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/main/src/facer.c.  
I need your help to check if including these headers would break the module for you or not. Reacting as Like to this post if it worked for you or Dislike if it didn't can be helpful.

Or better, can someone do a R&D to know more about these headers? 

```
#include <linux/uaccess.h>
#include <asm/uaccess.h>
```

fernandoalbino, 2022-12-02:
Helios 300 PH314-54-70hl .... working now!!! 

System Information
        Manufacturer: Acer
        Product Name: Predator PH315-54
        Version: V1.14
        Serial Number: *
        UUID: *
        Wake-up Type: Power Switch
--
Base Board Information
        Manufacturer: TGL
        Product Name: QX60_TLS
        Version: V1.14
        Serial Number: *
        Asset Tag: Type2 - Board Asset Tag
        Features:
        
_________

snowyoneill, 2022-12-27:
@JafarAkhondali the last time i tried to install this module i dont believe i need to mess around with the headers. I will try again and report back