JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #75: Anything for fedora

SAADAT-Abu, 2022-12-26:
HI..

Can I in some way use it on fedora 37?

Thanks a lot.
Best wishes.

x211321, 2022-12-26:
I have it running on fedora 37 but it is somewhat finicky and seems to break after every kernel update. 

As far as I know most of these problems are due to complications with SELinux.

Try running the following commands to register the kernel module with SELinux (after normal installation according to the project readme).
```
sudo semanage fcontext -a -t modules_object_t /opt/turbo-fan/src/facer.ko
sudo chcon -t modules_object_t /opt/turbo-fan/src/facer.ko
```
Afterwards Reboot.

Maybe a  [previews comment](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/52#issuecomment-1287947542) of mine or [this thread](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/49) about fedora 36 could be of interest as well. 

If it breaks after an update I found a complete reinstall of the module plus the two commands from above fix it most of the time.

asant, 2023-01-17:
Hi,
I have FC37 with SELinux enabled on enforcing mode, all with default settings, and I've never had any issues with it.
I also quickly wrote my own spec file for building an akmod src rpm, so that a kmod rpm gets built at every kernel update.
It could probably be improved, as I really didn't follow the RPM guidelines, but it still works fine for me. 
I'm attaching it, but in order to use it, you need to learn how to use rpmbuild and understand spec files. Also, if you have EFI enabled, you need to configure the OS so that it can sign your 3rd party modules.
[acer-predator-turbo-and-rgb-keyboard-linux-module-kmod.spec.txt](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/10434095/acer-predator-turbo-and-rgb-keyboard-linux-module-kmod.spec.txt)

asant, 2023-01-17:
Oh sorry, I forgot to mention that you also need to build another srpm too. I'm attaching the other spec file too.
[acer-predator-turbo-and-rgb-keyboard-linux-module.spec.txt](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/10434130/acer-predator-turbo-and-rgb-keyboard-linux-module.spec.txt)

droopy4096, 2023-08-03:
@asant I've taken your code and started wrapping it into something buildable... it's a work-in-progress but in case someone cares :+1: 

https://github.com/droopy4096/acer-predator-rpm

@asant feel free to reach out directly if license doesn't suit you

JafarAkhondali, 2023-08-03:
@droopy4096 When you are done, it would be nice to have a reference to your project in readme. You can also create a PR and merge the project if u want

asant, 2023-08-03:
hey guys, sorry, I had completely forgotten about this thread. I made a copr repository available, you can find it here together with other modules:
https://copr.fedorainfracloud.org/coprs/asan/acer-modules/
Those are the modules I used in order to configure my Nitro 5.
These packages use akmods so the modules get rebuilt every time you update the kernel. Of course you need to enable package signing if you don't disable uefi.

droopy4096, 2023-08-05:
@JafarAkhondali https://github.com/droopy4096/acer-predator-rpm is in "beta" quality for the moment (i.e. it compiles "out of the box" but not necessary functions as I just finished working on it and didn't have a chance to test). @asant provided F38 packages, my repo modifications of @asant work should allow folks with any Fedora version to compile packages for themselves. Mind you COPR should satisfy most, as long as packages for popular/current OS versions are available.