JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #77: Cannot install as systemd service on an515-55

ghtesting2020, 2022-12-27:
Hi thank you for making this software it is very cool! I bought an an515-55 for my son for Christmas but we are having trouble installing as systemd service. Only works as a one time install and need to do install every time restart.

According to the chart his model does not support turbo fan but whenever we try to run the systemd service install it says turbofan found, removing then it asks for luks encryption password which makes no sense. I have never had a systemd service ask for luks encryption password before only sudo password. Regardless when provide correct password or press tab it does not install properly. Also it says something about 
"./install_service.sh: line 44: rsync: command not found"

my son is using arch linux fully updated. Here is some output with a couple values xx out.

```
[ryan@acer acer-predator-turbo-and-rgb-keyboard-linux-module]$ sudo ./install_service.sh 
[sudo] password for ryan:  
[Mode: install] 
['turbo-fan' service is presented. Remove it.] 
Removed "/etc/systemd/system/multi-user.target.wants/turbo-fan.service". 
[Remove old data] 
removed directory '/opt/turbo-fan' 
[Create directories] 
[Copy new data] 
./install_service.sh: line 44: rsync: command not found 
[Create turbo-fan service] 
 Please enter passphrase for disk xxxxxxxxxxxx (xxxxxxxxxxxxxx): (press TAB fo
*********** 
Created symlink /etc/systemd/system/multi-user.target.wants/turbo-fan.service → /etc/systemd/system/turbo-fan.servi
ce. 
[ryan@acer acer-predator-turbo-and-rgb-keyboard-linux-module]$  

```

then if we try to run it via RGB gui or command line it says:

```
RGB Device /dev/acer-gkbbl-0 not available
```
but if we just do the normal install not as service then the gui or command line works without issue (but need to install every single time restart..)

x211321, 2022-12-28:
The install_service-script uses rsync to install the service files to /opt/turbo-fan.

To me it sounds as if the rsync command is missing on your setup. 

Please check the output of 
`whereis rsync`

in a terminal. It should show path where rsync is installed.

If no path is shown then rsync is not installed. 

Try installing it via
`packman -S rsync`

See also: https://wiki.archlinux.org/title/rsync

ghtesting2020, 2022-12-28:
> The install_service-script uses rsync to install the service files to /opt/turbo-fan.
> 
> To me it sounds as if the rsync command is missing on your setup.
> 
> Please check the output of `whereis rsync`
> 
> in a terminal.
> 
> It should show the path where rsync is installed.
> 
> If no path is shown then rsync is not installed.
> 
> Try installing it via `sudo packman -S rsync`
> 
> See also: https://wiki.archlinux.org/title/rsync

Yes it looks like that was it. I installed rsync now and will try again, but before I do is there any way to install the service without the turbo fan for those of us that don't have that support?

x211321, 2022-12-28:
The turbo fan part doesn't really matter. I'm running the module on a Nitro 5 which has no turbo fan as well. The naming scheme of files/folders/modules/etc. is somewhat inconsistent. If you follow the instruction from the readme the RGB module should work independent of any turbo fan functionality. 

FYI: I edited my last comment, might check it again just in case.

ghtesting2020, 2022-12-28:
> The turbo fan part doesn't really matter. I'm running the module on a Nitro 5 which has no turbo fan as well. The naming scheme of files/folders/modules/etc. is somewhat inconsistent. If you follow the instruction from the readme the RGB module should work independent of any turbo fan functionality.
> 
> Edit: turbo-fan in this case is simply the name of the service that handles loading the kernel module. When it states `['turbo-fan' service is presented. Remove it.] ` it only means that you had previously installed the service and are now reinstalling it thus the previous installation is removed.
> 
> The kernel module that is loaded by the 'turbo-fan' service handles all the custom acer functionality that this project provides (including RGB). The name 'turbo-fan' is just a little misleading and probably a holdover from an earlier state in development.
> 
> FYI: I edited my last comment, might check it again just in case.

It no longer asks for luks password now that I have rsync installed. Everything is good now. Thank you so much for help!

closed 2022-12-28 by ghtesting2020