JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #78: reverse engineering guide

ghtesting2020, 2022-12-28:
Hi i was wondering when the video guide will be available for
"As most of the contributions in this project are less related to the C code and as I don't own a Predator anymore, I'll soon release a video tutorial on how to reverse engineer other features of your laptop and implement it in this project."
it is incredible you were able to reverse engineer this and i am curious how you did it.

JafarAkhondali, 2023-01-01:
Hi,
Sorry I got few problems and couldn't record the video, but I'll definitely record it.  
Also I'm not sure publishing a tutorial on reverse engineering Acer software is legal or not.  
To get started, I used DotPeek to understand logic of PredatorSense(it was developed in C# so DotPeek is really useful here).  
I Also Used WMI Explorer and Process manager to monitor and detect WMI changes(and play with PredatorSense at the same time). Once you know what WMI functions are used, and what are the parameters, you can play with them programmatically in Windows, but not Linux. Because Windows allows WMI calls in user-space, but in order to use WMI calls in Linux, you'll have to create a driver as a bridge to communicate with kernel-space.  
Some challenges might be finding [WMI ID](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/8c2b628f83bb7d0321db4e7e1e4ea7a34f88fe1f/src/facer.c#L82) and [WMI Function Ids](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/8c2b628f83bb7d0321db4e7e1e4ea7a34f88fe1f/src/facer.c#L68) which can be detected by WMI Explorer or de-compiling ACPI table(WMI Explorer is much easier but it requires Windows).

JafarAkhondali, 2023-02-19:
I just started recording the guide, it's my first time recording a tutorial in English so I hope this would work for
https://www.youtube.com/watch?v=97-WNhUmoig&list=PLv2kA4LxAI4Dq2ic_hU9bdvxIzoz5SzBr

closed 2023-02-19 by JafarAkhondali