JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #79: PH315-55 support

DS3a, 2023-01-02:
# Model: ```System Information
	Manufacturer: Acer
	Product Name: Predator PH315-55
	Version: V1.05
Base Board Information
	Manufacturer: ADL
	Product Name: Evoque_ADH
	Version: V1.05
```

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1 

### RGB keyboard works?  No
### Turbo button turn on fans? No
### Turbo button turn on LED? o
### Turbo button activates overclock? No

I am willing to contribute to making these features work. I don't really care for RGB or clocking. I just want control over the fan speeds, although I don't mind writing code for all. In #78 you mentioned that you won't be able to make a public tutorial on reverse engineering. How do I get started with it then?

JafarAkhondali, 2023-01-02:
You can get started by Decompiling Acer Predator Sense using DotPeek. From there, you'll need to find which WMI functions would allow controlling each fan speed, and what other WMI functions provide. For example, some WMI functions may give fan speed, etc.

abhijeet-singh800, 2023-01-22:
I have a Predator Helios 300 (PH315-55) . The keyboard RGB is responding to the repository. However, Acer added a light bar on the laptop's front edge. I was wondering if the WMI of the keyboard and light bar is the same and if there is any way to switch off the light bar but not the keyboard.

JafarAkhondali, 2023-02-17:
@abhijeet-singh800 I believe the light-bar would have similar mechanism to the keyboard, but I can't help with the implementation as I don't have a model similar to you. I'm starting tutorials for the implementations so you (or other people) could help improve this project.

JafarAkhondali, 2023-02-19:
@DS3a 
I just started recording the guide on [youtube](https://www.youtube.com/watch?v=97-WNhUmoig&list=PLv2kA4LxAI4Dq2ic_hU9bdvxIzoz5SzBr). It's my first time recording a tutorial in English so I hope this would work.

marvinhrivera, 2023-05-27:
I don't think this is going to help in anything as I'm not skilled in coding but rgb keyboard works for me, turbo mode didn't work but I just tried adding the two following blocks of code and when I pressed the turbo button it actually switch both fans at max speed (turbo) but I couldn't turn it off even after reboot until I uninstalled the module. Is there anything else I could try since it seems to be turning turbo mode on?

 
`static struct quirk_entry quirk_acer_predator_ph315_55 = {
		.turbo = 1,
		.cpu_fans = 1,
		.gpu_fans = 1,
};`

```
		{
				.callback = dmi_matched,
				.ident = "Acer Predator PH315-55",
				.matches = {
						DMI_MATCH(DMI_SYS_VENDOR, "Acer"),
						DMI_MATCH(DMI_PRODUCT_NAME, "Predator PH315-55"),
				},
				.driver_data = &quirk_acer_predator_ph315_55,
		},
```

BerekHalfhand, 2023-06-18:
@JafarAkhondali hi, i also have a PH315-55 and have done what @marvinhrivera did with the same result. It's been years since i wrote a string in C, but i'm a programmer and can collaborate with you on making your code work with PH315-55.
From what i can tell, the function `acer_toggle_turbo` does not get called in the event of pressing the turbo btn when it's already on.
Either
`	if (ACPI_FAILURE(WMID_gaming_get_u64(&turbo_led_state, ACER_CAP_TURBO_LED)))
		return -1;
`
or
`
			if (return_value.key_num == 0x4)
`
seem to be the culprit, but to know what i'd need a way to debug those pieces. Would you explain how do i run this code with debugging possibilities?

BerekHalfhand, 2023-06-18:
By the way, after this incident with the fans stuck in maximum power, my laptop had an issue powering off, and the turbo mode appeared to be broken even after uninstall. [PredatorNonSense](https://github.com/kphanipavan/PredatorNonSense) was getting turbo mode to work before i tried that move with adding PH315-55 into the code.
So what ended up solving the issue was to run [refresh.sh](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/main/refresh.sh) as root. It doesn't seem to be documented, so hope this comment helps somebody

JafarAkhondali, 2023-06-20:
@BerekHalfhand I used the state of the Led below turbo button to detect whether turbo button is on or off. This might be a bad idea but at that time this was the fastest method available to me :) For debugging, I believe you can check dmesg logs and print debugging messages with pr_err function. Again, this also might sound so bad but it's all I can remember when I developed this project.