JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #80: Changing keyboard brightness somehow affects the display brightness

Evan-aja, 2023-01-11:
## Model: `sudo dmidecode | grep "Product Name" -B 2 -A 4`

```javascript
System Information
	Manufacturer: Acer
	Product Name: Nitro AN515-57
	Version: V1.19
	Wake-up Type: Power Switch

Base Board Information
	Manufacturer: TGL
	Product Name: Scala_TLS
	Version: V1.19
	Asset Tag: Type2 - Board Asset Tag
```

### Count of keyboard RGB zones: 4

### Count of Cpu Fans: 1
### Count of Gpu Fans: 1 

### RGB keyboard works?  Yes
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No

I have found a behavior on my Nitro 5 where whenever I increase the keyboard's brightness, it also calls the display's decrease brightness button.

When I adjust the display brightness to normal, the display's decrease brightness button cease to function. this was tested on `Gnome` and `KDE`, `X11` and `Wayland` on vanilla `ArchLinux`. this issue persist until the next boot of the system, until the button is pressed again that is.

[image] 

nothing appeared on terminal nor systemd log. this issue has long existed even before my merge request on #74.

Kapitoha, 2023-01-13:
Hi. I have the same behavior on my PH317-53. But I thought it was because I replaced my keyboard :smile:

snowyoneill, 2023-01-17:
Adding reference to previously discovered bug https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/24

Evan-aja, 2023-02-06:
A workaround that I found is before changing keyboard brightness
1. switch to an unused `TTY` session with `ctrl+alt+f{1-12}`
1. change the keyboard brightness arbitrarily
1. return to desktop's `TTY` session (can be `f1`, `f2`, or `f7`, try whichever works)
1. now it is possible to change the keyboard brightness without affecting the screen's brightness, at least until the bug is fixed
1. do this on every reboot

Evan-aja, 2023-04-14:
On KDE Plasma 5.27 with kernel 6.2.10, although the screen brightness changed, the brightness down function still works.

Sinfolke, 2024-10-17:
Also have the exactly same issue on Acer Nitro 5 AN515-58. After brightness increase it reduces screen brightness to zero. I might change it from control panel but any click then returns it back. 
Helped to reload cinnamon (possibly linux mint only solution): ctrl + alt + esc. This issue disappears until next reboot.