JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #81: Reporting as working on AN515-57-70MT

asant, 2023-01-17:
Hi, I just wanted to report this module as working on Aspire AN515-57-70MT (TGL Scala_TLS board V1.19 with Type2 asset, this model lacks a turbo button) since December commits.  I'm using it on Fedora 37, kernel 6.1.6-200.
Somehow, it also makes another 3rd module for fan control work, namely acpi_ec made by musikid.

I was also wondering: why it is necessary to unload the default acer_wmi before loading this module? Are there any chances to make both coexist?

JafarAkhondali, 2023-01-17:
Hi, thank you for the report.
We have to remove `acer_wmi` module because the current code is implemented inside the actual `acer_wmi` module. I believe it wasn't possible to use multiple modules for same WMI_GUIDs. So I had to replce the current acer_wmi.c file with mine.
It's better to apply patches, but I didn't have time to do it.

asant, 2023-01-18:
Thanks for your reply, the information and the great work @JafarAkhondali

closed 2023-01-18 by asant