JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #85: support for kernel 6.2.x?

groopcd, 2023-02-28:
unable to compile with kernel 6.2.x

make -C /lib/modules/6.2.1-060201-generic/build M=/home/groo/devel/predator/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-6.2.1-060201-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-9ubuntu1) 12.2.0
  You are using:           gcc-12 (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  CC [M]  /home/groo/devel/predator/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/groo/devel/predator/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘gaming_kbbl_cdev_init’:
/home/groo/devel/predator/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:2019:37: error: assignment to ‘int (*)(const struct device *, struct kobj_uevent_env *)’ from incompatible pointer type ‘int (*)(struct device *, struct kobj_uevent_env *)’ [-Werror=incompatible-pointer-types]
 2019 |         gkbbl_dev_class->dev_uevent = gkbbl_dev_uevent;
      |                                     ^
/home/groo/devel/predator/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘gaming_kbbl_static_cdev_init’:
/home/groo/devel/predator/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:2123:44: error: assignment to ‘int (*)(const struct device *, struct kobj_uevent_env *)’ from incompatible pointer type ‘int (*)(struct device *, struct kobj_uevent_env *)’ [-Werror=incompatible-pointer-types]
 2123 |         gkbbl_static_dev_class->dev_uevent = gkbbl_static_dev_uevent;
      |                                            ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:252: /home/groo/devel/predator/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[1]: *** [Makefile:2027: /home/groo/devel/predator/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.2.1-060201-generic'
make: *** [Makefile:9: default] Error 2
rmmod: ERROR: Module acer_wmi is not currently loaded
insmod: ERROR: could not load module src/facer.ko: No such file or directory

thank you

Evan-aja, 2023-02-28:
### It is fixed on PR #86

groopcd, 2023-02-28:
thanks, its fixed now.

closed 2023-02-28 by groopcd