JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #87: Nitro AN517-41

F-U-B-AR, 2023-03-02:
## Model: Nitro AN517-41
### Count of keyboard RGB zones: 4
### RGB keyboard works?  No
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No
### OS: Kubuntu 22.04.2 LTS

I Know that the model can work: https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/70

_I feel like I am missing something obvious!_

When I run the python script I get:

> "Traceback (most recent call last):
>   File "/home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/facer_rgb.py", line 203, in <module>
>   with open(CHARACTER_DEVICE, 'wb') as cd:
> PermissionError: [Errno 13] Permission denied: '/dev/acer-gkbbl-0'"

with sudo nothing happens:
I have checked the folder /dev and there isnt a acer-gkbbl-0 folder.
[I ran the installation (reboot) instructions without issue]

Any help would be appreciated,
Cheers

F-U-B-AR, 2023-03-04:
I tried following the fix to the known issue here: https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/4#issuecomment-905486393
Didnt seem to work... (_the keyboard light remains the same colour_) 
I also checked secure boot in the BIOS, its definitely **off**.

I then uninstalled, and tried to reinstall and got this output on the terminal: 

make -C /lib/modules/5.19.0-35-generic/build M=/home/nova/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-35-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0                                                         
  CC [M]  /home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o                                           
/home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘gaming_kbbl_cdev_init’:
/home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:2019:37: error: assignment to ‘int (*)(struct device *, struct kobj_uevent_env *)’ from incompatible pointer type ‘int (*)(const struct device *, struct kobj_uevent_env *)’ [-Werror=incompatible-pointer-types]
 2019 |         gkbbl_dev_class->dev_uevent = gkbbl_dev_uevent;
      |                                     ^
/home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘gaming_kbbl_static_cdev_init’:
/home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:2123:44: error: assignment to ‘int (*)(struct device *, struct kobj_uevent_env *)’ from incompatible pointer type ‘int (*)(const struct device *, struct kobj_uevent_env *)’ [-Werror=incompatible-pointer-types]
 2123 |         gkbbl_static_dev_class->dev_uevent = gkbbl_static_dev_uevent;
      |                                            ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:257: /home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[1]: *** [Makefile:1850: /home/user/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-35-generic'
make: *** [Makefile:9: default] Error 2
insmod: ERROR: could not load module src/facer.ko: No such file or directory

@JafarAkhondali Have you got any idea on what could be causing the issue please?

JMans15, 2023-03-04:
I didn't use the systemd installation because I don't like to have too much daemons running in the background when it's not truly needed, is it working with the 'normal' install script?

If you installed the service you should be able to see the logs of the latter through ```systemctl status turbo-fan.service```

F-U-B-AR, 2023-03-04:
I'm afraid I tried that, it isn't work with the normal install script either..
On running the `systemctl status turbo-fan.service` script I recieve:
```
○ turbo-fan.service - Enables turbo button
     Loaded: loaded (/etc/systemd/system/turbo-fan.service; disabled; vendor preset: enabled)
     Active: inactive (dead)
```
I believe its do with the `": Module facer is not currently loaded"` error, which is happening due to
`src/facer.c:2019:37: error: assignment to ‘int (*)(struct device *, struct kobj_uevent_env *)’ from incompatible pointer type ‘int (*)(const struct device *, struct kobj_uevent_env *)’ [-Werror=incompatible-pointer-types]`

This is a beyond my abilities but I shall keep trying!

F-U-B-AR, 2023-03-04:
_Update -_ I tried reproducing this error and ran the installer on a live version of MX Linux:

the same `ERROR: Module facer is not currently loaded`
and `/src/facer.c:2019:37: error: assignment to ‘int (*)(struct device *, struct kobj_uevent_env *)’ from incompatible pointer type ‘int ` / `[-Werror=incompatible-pointer-types]`

JMans15, 2023-03-05:
Your service seems to be disabled, try ```sudo systemctl enable turbo-fan.service```
(Enabling a service starts it automatically during each boot)
Also do ```sudo systemctl start turbo-fan.service``` to try to run it, you should see logs appear under the service informations (systemctl status)

The error coming from C is likely due to a compiler mismatch (C has no fixed semantic, two compilers may produce two different programs from the same code), try to find where to get ```x86_64-linux-gnu-gcc``` (I don't have access to a Ubuntu machine right now)

F-U-B-AR, 2023-03-07:
### **Solved!**

I believe I have fixed the issue, and you can replicate the _bug_   _/_ _user error_ ...
If you run the install **twice** (_I must of done by accident_) :

**1.** `cd "acer-predator-turbo-and-rgb-keyboard-linux-module"`
**2.** `chmod +x ./*.sh`
**3.** `sudo ./install.sh`

**4.** close Terminal
**5.** re-open Terminal

**6.** `cd "acer-predator-turbo-and-rgb-keyboard-linux-module"`
**7.** `chmod +x ./*.sh`
**8.** `sudo ./install.sh`
**9.** `./facer_rgb.py -m 3 -s 5 -b 100`

You get errors such as: 
```
ERROR: Module acer_wmi is not currently loaded                        
ERROR: could not insert module src/facer.ko
Permission denied: '/dev/acer-gkbbl-0'
```
This would normally be fixed on a reboot  but whats interesting is KDE/Kubuntu has a persisting desktop session enabled by default. _(System Settings / Startup & Shutdown / Desktop Session / "when logging in")_ 
So I was effectively booting into a bugged system. 

(_I still cant understand why it was happening on a fresh install of MX Linux but I am just happy that I have fixed it._)

**A big thank you to @JMans15 for helping me try to solve this and a big thank you to @JafarAkhondali for creating this project.**

closed 2023-03-07 by F-U-B-AR