JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #92: PT516-52s RGB working, Turbo not working

NiklasWilson, 2023-03-25:
PT516-52s
This laptop seems to have 3 RGB zones. RGB commands are working for zones 1-3.
The turbo button is not working though.

sudo dmidecode | grep "Product Name" -B 2 -A 4
System Information
	Manufacturer: Acer
	Product Name: Predator PT516-52s
	Version: V1.04
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: ADL
	Product Name: Volante_ADH
	Version: V1.04
	Asset Tag: Type2 - Board Asset Tag
	Features:

JafarAkhondali, 2023-06-17:
Hi, thanks for informing! Does your laptop have a turbo button?

quentingllmt, 2023-07-01:
Hi @NiklasWilson, 

I have installed ubuntu 22.04 on this computer but I'm facing an issue where the screen (backlight) starts flickering (rapid flashes) after a suspend / sleep on low brightness. The screen does not flashes on first boot or on full brightness.

Do you have the same problem ?

NiklasWilson, 2023-07-01:
> Hi, thanks for informing! Does your laptop have a turbo button?

Yes it has a turbo button. After awhile I figured out the issue was that Linux didn’t see that my laptop had any fans. I gave up trying to resolve the issue and went back to windows.

NiklasWilson, 2023-07-01:
> Hi @NiklasWilson,
> 
> I have installed ubuntu 22.04 on this computer but I'm facing an issue where the screen (backlight) starts flickering (rapid flashes) after a suspend / sleep on low brightness. The screen does not flashes on first boot or on full brightness.
> 
> Do you have the same problem ?

quentingllmt, 2023-07-02:
> > Hi @NiklasWilson,
> > I have installed ubuntu 22.04 on this computer but I'm facing an issue where the screen (backlight) starts flickering (rapid flashes) after a suspend / sleep on low brightness. The screen does not flashes on first boot or on full brightness.
> > Do you have the same problem ?
> 
> i was using Linux mint, and did not have any of these issues.

Is tried linux mint (Cinnamon) and I see the same issue..

commit 4379a9ea (2023-07-27): [Add new models for turbo mode]
+ Add support for PT314-52S model - Fix #108
+ Add support for PT516-52s model - Fix #92
+ Add support for PH315-55  model - Fix #105
closed 2023-07-27 by JafarAkhondali