JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #93: Error when installing on predator triton 500, ubuntu 22.04

ccunarro, 2023-04-08:
"kobject_add_internal failed for acer-gkbbl with -EEXIST, don't try to register things with the same name in the same directory."

Here are the full logs when running the command.

`carlos@predator:~/Tools/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo bash install.sh 
make -C /lib/modules/5.19.0-38-generic/build M=/home/carlos/Tools/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-38-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-38-generic'
[ 2544.805188] RDX: 0000000000000000 RSI: 0000564fdfd98cd2 RDI: 0000000000000003
[ 2544.805189] RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000000
[ 2544.805191] R10: 0000000000000003 R11: 0000000000000246 R12: 0000564fdfd98cd2
[ 2544.805192] R13: 0000564fe060e780 R14: 0000564fdfd97888 R15: 0000564fe060e8f0
[ 2544.805194]  </TASK>
[ 2544.805210] kobject_add_internal failed for acer-gkbbl with -EEXIST, don't try to register things with the same name in the same directory.
[ 2544.814222] nouveau 0000:01:00.0: disp: error 00000001
[ 2544.814507] input: Acer WMI hotkeys as /devices/virtual/input/input52
[ 2544.826578] nouveau 0000:01:00.0: disp: error 00000001
[ 2544.834234] nouveau 0000:01:00.0: disp: error 00000001
[*] Done
`

Any ideas of what could be going on?

Evan-aja, 2023-04-14:
Have you ever installed the module prior to this instance? if yes maybe try to uninstall then install the module. everything works fine for me on kernel 6.2.10.

JafarAkhondali, 2023-05-10:
Make sure you didn't used install service at first. Uninstall everything, reboot, then reinstall again using the service install script

IllusionMan1212, 2023-10-06:
This should be fixed by #121

closed 2023-10-07 by JafarAkhondali