JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #94: Does not work with PH517-61

TheFeshy, 2023-05-09:
The module compiles fine on Arch, but attempting to insmod it gives  `ERROR: could not insert module src/facer.ko: Input/output error`

The relevant part of dmesg is:

`[ 1297.153883] facer: loading out-of-tree module taints kernel.
[ 1297.153982] facer: module verification failed: signature and/or required key missing - tainting kernel
[ 1297.155612] facer: Acer Laptop ACPI-WMI Extras
[ 1297.155653] facer: Function bitmap for Communication Button: 0x801`

I suspect the problem is that this model has no turbo button; instead it has an array of six customize-able buttons along the top. ([Here](https://diit.cz/sites/default/files/helios35.jpg) is an image of them)

# Model: `sudo dmidecode | grep "Product Name" -B 2 -A 4`

`System Information
	Manufacturer: Acer
	Product Name: Predator PH517-61
	Version: V1.09
	Serial Number: NHQ3GXXXXXXXXXXXXXXXXX
	UUID: XXXXXXXXXXXXXXXXXXXXXXXX
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: PRS
	Product Name: Cayman_PRS
	Version: V1.09
	Serial Number: NBQ3GXXXXXXXXXXXXX
	Asset Tag: Type2 - Board Asset Tag
	Features:`

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1  

### RGB keyboard works?  Yes\No

No

### Turbo button turn on fans?  Yes\No
### Turbo button turn on LED? Yes\No
### Turbo button activates overclock? Yes\No

No turbo button present

I'm only a little interested in getting the RGB to work; what I would really like is the ability to turn the fans to a higher mode.  The default setting just isn't sufficient for cooling the laptop during gaming or especially while running stable diffusion!

JafarAkhondali, 2023-05-09:
Hi, try disabling secure boot and apparmor, then reinstall the module again

TheFeshy, 2023-05-09:
Secure boot and apparmor were already disabled in the first attempt, sorry for not mentioning that.

`$> mokutil --sb-state`                                                                                                                                                                    
`SecureBoot disabled`
`Platform is in Setup Mode`

`sudo systemctl status apparmor `
`○ apparmor.service - Load AppArmor profiles`
`     Loaded: loaded (/usr/lib/systemd/system/apparmor.service; disabled; preset: disabled)`
`     Active: inactive (dead)`

TheFeshy, 2023-05-09:
I found the problem - I still had acer_wmi loaded.  Removing that allowed the module to install.

However, I have limited functionality using the python script.  I can control red and green, in static mode, for all zones.  No other functionality (blue set to 0 or 255 makes no difference, brightness makes no difference, and none of the other modes like breath or wave do anything.)

And of course, there's no turbo button to test that functionality.  Is there a way to test fan speeds?  Fan speed increase is what I would most like to use.

JafarAkhondali, 2023-05-09:
Hmm that's weird ... maybe the order of bytes in payload is different for your laptop? but I've never hear such a problem. So first try to boot into Windows, play with colors, then test again in Linux. If you followed the install script the module should not get loaded.
You can try [nbfc](https://github.com/hirschmann/nbfc) for fan control, it supported my model but not sure about yours. If you manage to find the turbo or similar key, maybe you can try using that key instead of the turbo key by modifying some of the C code.

TheFeshy, 2023-05-09:
Okay, I tried in windows and it looks like somewhere in the last few years my blue lights have stopped working, and I didn't notice because I am always in linux lol.

nbfc was unable to help, unfortunately - there just wasn't any useful information in the DSDT (actually the DSDT was blank; but there were quite a few SSDT's.)  Unfortunately the EC literally only had one byte of memory defined.

I'll see if any of these extra keys show up somewhere that I can use them to trigger the code in the module.

TheFeshy, 2023-05-10:
As a brute-force test, I added the fan turbo function to the module init, and fan auto to the module unload.  The methods work fine with this laptop.  Unfortunately the lack of a turbo button (and you use it's LED as an internal state for turbo  mode) mean it's a bit less simple to use another button.  But for now at least I have a crude way to turn the fans to max to get some stable diffusion done and test a few other laptop issues!

JafarAkhondali, 2023-05-10:
Good to hear that. I think I used LED for state because either that's what Acer already did or I wanted to check state cross-OS. You can simply use a variable in driver or even a Char device to control that instead of LED. If you are modified the C code already, make sure to use correct count for CPU fans and GPU fans. Most models have 1 for each, but some have 2. https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/blob/f9deb86d791d5727a74f149d188dc593373abd4f/src/facer.c#LL487C1-L491C3